prusa3d/PrusaSlicer #9132: Brim where needed only

Titlest, 2022-11-06:
_**Is your feature request related to a problem? Please describe.**_

**FEATURE REQUEST**
Sometimes when I do large prints that are filling up the print bed and also have some complex shapes and together with the usage of "brim" it is a big problem when cleaning parts after printing and it also is time-consuming. I need the brim on the parts so that they don't warp and detach from the build plate. sometimes even a 10 mm brim is needed.  
Imagine a print speed of 15mm/s on the first layer and you have a complex shape that has a lot of corners but only a few of them need brims to be able to stick to the print bed. One round trip for the brim can be as an example, 500mm which will have a print time of 33 seconds (500mm/15mm's = 33) and that will be 0.5mm wide per revolution. 10mm brim will then have 20 revolutions which will be a total of 11 minutes ( 33sec'rev * 20 rev = 660sec / 60 sec=11min ) just for printing the brim. It also uses a lot more plastic than needed.

_**Describe the solution you'd like**_
For me, the solution is to just use a brim on the corners of a part where I see the problems for warping may happen.
To save material and print time on large prints I am today modeling my own brim where I see that it is needed. "see picture"

_**Describe how it would work**_
Where we have the "Paint on support" and "Seem painting" I would like to see "place brim"
When active you will be able to set the size of your brim to a radius and then click on corners from the model that is on the build plate. This will add a brim to the corner with the specified radius.

_**Describe alternatives you've considered**_
Keep on modeling the brims manually in CAD and do not use the brim function in the Slicer

[image]

Titlest, 2022-11-06:
Didn't see the duplicate at first
https://github.com/prusa3d/PrusaSlicer/issues/8283

closed 2022-11-06 by Titlest

Snuff1eupagus, 2024-04-26:
Try this, it may enlighten you.

https://www.youtube.com/watch?v=dhsbMmosbLU&t=4s

Enjoy