prusa3d/PrusaSlicer #9138: temporarily disable background calculation over user interface

M-K3y, 2022-11-07:
Hello, 

I almost exclusively use the background calculation for slicing, unfortunately it sometimes happens that with large or complex objects when creating supports or when aligning several individual parts, the background calculation takes quite a long time with each step, with a step that is repeated 10 times must or more, as currently the best solution is to disable this via Config => Settings => Background Calculation and then enable it again afterwards, 

maybe a good optimization would be to create a small button, e.g. between the margin selection button and the cleaning volume settings, which can be used to turn the background calculation on and off directly from the user interface, or temporarily disable it until the next program start or manual slicing to generate gcode

foreachthing, 2022-11-07:
> maybe a good optimization would be to create a small button

Or, just a shortcut would be nice.

gudvinr, 2022-11-09:
That's duplicate of #8672. Despite brush on support in the name, core issue is the same.
Underlying problem seem to be coming from the fact that PS triggers processing after each change and it doesn't delay calculation while user still interacts with UI and.  
Shortcut or button to manually turn that off is a mere workaround here.

closed 2026-09-29 by prusaslicer-bot[bot]