prusa3d/PrusaSlicer #9139: Perimeter and modifier fiddles to improve durability of model

DejfRCR, 2022-11-07:
Hello,
1) fuzzy skin as it is does not have any effect on object durability and I think there is a way to do so without much hassle.
Principle is that perimeters are stronger if they are not placed exactly one on another, thus the slight randomness in their placement would help making stronger prints - but fuzzy skin would have to affect all perimeters for that not just the external one.

2) Without fuzzy skin there would be another great way: make modifiers layer bound and repetitive, so if every second layer has one more perimeter while perimeters get thinner, the resulting print would get more durable. Also this would allow things like do solid infill on two layers every 20 layers and so on. One layer of solid infill does not make much durability, its cool to support partial solid infills on low underlayng infills that are generally incapable of holding arbitrary bridging solids. The repetitive ability would make it much easier to set things and whats more important: change them. Changing 20 modifier slabs is insane...

murk-sy, 2022-11-09:
Fuzzy skin is a bit of a hack that changes perimeters after they were generated and it's generally used more more either for aesthetic or gripping reasons. It's not really a good way to add strength improvements.

As for offset perimeters, while that would *maybe* improve layer adhesion, the devs will want to see at least a proof of concept before they commit to implementation.

A relatively simple way to do that is to make a cylinder model where every second layer is inset 0.2 mm, so the extrusion on the layer above happens exactly in the middle of perimeters. It would require some careful measurements to make valid comparisons vs. standard printing, but it would serve as a baseline if the feature is worth implementing. It would require printing some perimeters thinner or thicker (likely the second outermost one), so the benefits would probably not show until a much larger number of perimeters is used with more engineering grade materials.

DejfRCR, 2022-11-09:
I tried these, both. Creation of models to try fuzzy is easy, but it does not prove anything about the perimeter changes. I dont know why fuzzy should not be changed to lead to better durability - I get that 90% of home printers aim only at aesthetic side of printing, yet this would not harm them. I dont have any serious measuring equipment, so the only result I can give is that durability rises by 20-50%, the resulting piece is clearly stiffer in hand.

The only way to show something real in the perimeter area is to go thru the modifier hell of layer by layer changes, only that way one could have two identical pieces containing the same width of external wall. Other way around could be just try to write a few lines of code to create these modifiers - but thats currently beyond my abilities.
3/4 perimeters is well enough and yes, it would help nylons and pla more than petg. But even with petg one could feel the difference.

I just made a sugestion, if developers are not interested, I would not push it. I tried in other projects and persuading people was never worth the time. It is based on polymer science and I guess Prusa has some well educated folks in that area - so they can theoretically prove that concept. Well, simply making perimeters far bigger than nozzle makes the model more durable too, also based on basics from polymer books I got. But you never find that in any printers handbook... Maybe because no such guide is written by polymer chemists and as far as I have spoken to people I was the only one finding out that school knowledge of polymers is not enough for good printing - thus I bough two polymer books from Czech VSCHT and from time to time I study deeper.

And btw cloning of modifiers with bound setting could be usefull in many other areas for many other hacks - as long as one has to change dozens of them with every new idea, one simply gives up on the idea. The first example that comes to my mind are Duplo railways that are made with a few modifier objects of two kinds, the STL itself dont print well. To propagate one chage to all instances is anoyance even with those few.

closed 2022-11-09 by DejfRCR