prusa3d/PrusaSlicer #9140: Import recent object STL, 3MF, ...

BMacours, 2022-11-07:
I'm making intensive use of File/Import/Import STL... Since I'm always using the same file names for export from FreeCAD (whatever the modeling tool) and import in PS,  having to always select the same files for import in PS (via file dialog) is really annoying and time consuming.

Please could you consider adding a menu entry under File/Import showing the 9 recently imported object files.

Thanks in advance,
Bernard

Forum link : [https://forum.prusa3d.com/forum/prusaslicer/import-recent-stl-obj-amf/#post-625895]

gordo3di, 2022-11-10:
Not sure if this helps, but since I'm constantly iterating objects as I'm prototyping I also save them with the same filename. I simply select the model in PS, right click and select "reload from disk" or sometimes "replace with stl". But I still think the feature is a good idea.

BMacours, 2022-11-14:
Yes indeed, "Reload from disk" is a nice feature when you are focusing on a few objects and when you are sure to use the same temporary filename for each one over the time. Moreover "Reload from disk" does not work great with amf files containing several objects.

BMacours, 2026-09-22:
Still in my wish list, thx.