prusa3d/PrusaSlicer #9147: Filament Profiles Selection crops off Printer names

VH-fixer, 2022-11-11:
### Description of the bug

When using the Filament Profile selection, the Printer name columns are not adjustable.  The result is the printer names are cropped and I cannot tell the difference between any of the MMUS2 configurations. 

### Project file & How to reproduce

Run Prusa Slicer
Open the configuration wizard
Have a MK3s+ with MMU2S configured.
Select Filaments

I cannot discern which MMU2S printer options are in the list I only see "Original Pursa i3 MK3 & MK3S+ MMU2S"  The rest of the name is cropped. 

[slicer-bug1.zip](https://github.com/prusa3d/PrusaSlicer/files/9991652/slicer-bug1.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0 Windows64

### Operating system

Windows 11 22H2

### Printer model

MK3s+ with MMU2S

DragonFinlay, 2022-11-11:
This dependent on the resolution of your monitor. If the UI has too little space, the sidebar isn't wide enough to show the full names. A workaround would be to select the profile, then hover the mouse over the dropdown to see the full name, but that's of course not ideal.

cjboco, 2025-12-29:
Even with a "large" monitor, this is still annoying. It expands the "Profile" column, but won't expand the "Printer Column". Makes it hard for me, with an MMU and different nozzle sizes. Even selecting the printer, doesn't show what it is... 

[image]

closed 2026-09-29 by prusaslicer-bot[bot]