prusa3d/PrusaSlicer #9150: Being able to set maximum line width for Arachne engine

Mike8-I, 2022-11-12:
In the advanced settings it is possible to set the nominal width for perimeters and the minimum feature width ( absolute or relative to the nozzle diameter).
It is absolute not clear what is the maximum feature ( perimeter) width. Also the Prusa support seems to be guessing.
It should be possible to set the maximum perimeter width.
E.g.: with my diamond back nozzle of 0.4mm it's possible to properly print lines from 0.25mm to 1.4mm with PLA ( tested).
I would like to set a nominal line width of 0.8mm ( possible), a minimum line width of 0.25mm (possible) and a upper limit of 1.4mm (not possible).

dougemon, 2022-12-01:
Hello Prusa Team,
I would also like to see a maximum line width for Arachne.
Thanks

xiphmont, 2022-12-30:
Maximum line width for Arachne is already determined by the other parameters; if Arachne can fit another bead without violating any of its contraints, it will.  If it can't, it won't.  That determines the maximum extrusion width. 

Setting a hard max width above the natural value will have no effect.  Setting it below will just cause Arachne to have to violate some other setting.

dougemon, 2023-08-14:
Hi again,
After tweaking all the available Arachne values in different ways, I am still ending up with lines that are too thick and don't extrude correctly. When printing somewhat small text elements, I would really appreciate a setting for maximum feature size.

Mike8-I, 2023-08-19:
> 

100% aggreed.

carlosjln, 2023-09-21:
In general the Arachne settings kinda work, but not when you need to fine tune things.

wikolii, 2025-04-24:
> Maximum line width for Arachne is already determined by the other parameters; if Arachne can fit another bead without violating any of its contraints, it will. If it can't, it won't. That determines the maximum extrusion width.
> 
> Setting a hard max width above the natural value will have no effect. Setting it below will just cause Arachne to have to violate some other setting.
> 
> What _might_ be nice is to compute/show the maximum extrusion width that results from the other constraints instead of just checking extrusion width after a wait.

Thanks for the info, but where can you see the maximum extrusion width being calculated?
All I see is minimum perimeter width and feature size but nothing so set or at least check the maximum.
If we knew how is this calculated, we could change other settings to accommodate for the desired max width.

Thanks!

neophyl, 2025-04-24:
They are probably slicing and then changing the preview legend selection to show width.  That way you can visually see which sections are printed with a wider extrusion.

dougemon, 2026-09-03:
This functionality has not been added to PrusaSlicer 3.0.0 alpha. Maximum line width should be directly controllable, as all too often the max that the arachne generator thinks is okay leads to gaps and underextrusion for lines that are too thick.