prusa3d/PrusaSlicer #9152: unneeded supports

SteWhite, 2022-11-13:
### Description of the bug

you can see the unneeded supports
[image]
[image]

### Project file & How to reproduce

[attacco tubo flessibile aspirapolvere.zip](https://github.com/prusa3d/PrusaSlicer/files/9996903/attacco.tubo.flessibile.aspirapolvere.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Windows 10 PRO x64

### Printer model

MK3S

neophyl, 2022-11-13:
Set the first layer expansion to zero too

DragonFinlay, 2022-11-15:
Very similar to #9087, with the difference that you need to adjust the closing radius AND the first layer expansion to get rid of the supports entirely.
The issue with the first layer having a support layer is somethine we should address though, so I'll keep the other issue open and close this one.

closed 2022-11-15 by DragonFinlay