prusa3d/PrusaSlicer #9153: Exporting STL damages model

F4r0kh, 2022-11-13:
### Description of the bug

I have downloaded [THIS](https://www.printables.com/cs/model/315674-f1-tires-pencilpen-holder) model designed for MMU and separated the parts so that they could be printed with multiple colours on a regular MK3 and wanted to upload the separate models back to Printables for others. When exporting the STLs one of the parts gets damaged and when your re-import it back to the PrusaSlicer it is missing parts and will not print. I'm running the Prusaslicer 2.5.0. On the screenshot you can see the original model (orange) and the one which was exported to STL and imported back to the slicer (green).

[image]

### Project file & How to reproduce

[F1_tyres_slick.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/9997079/F1_tyres_slick.3mf.zip)

In the right bar when the model is selected, right click - export as STL, then add the exported STL back into the slicer.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

MacOS Monterey 12.6.1

### Printer model

Prusa MK3S+

DragonFinlay, 2022-11-17:
It's an issue with the model. Fixing it with NetFabb solved the issue (it says there's 1 open edge).

On the left the exported STL before fixing, on the right after fixing:

[image]

F4r0kh, 2022-11-18:
Thanks for the reply, I have noticed the error reported in the model, but I don't have the "fix through NetFab" option in the slicer, I tried to google why and found out that this feature is only available on Windows 10, is it true? Are you planning to emable it on MacOS? Thanks very much for your help ;)

Best regards, Ruslan

[image]

DragonFinlay, 2022-11-18:
Ah, right. I didn't see you're on MacOS. The NetFabb service is provided by Microsoft, so it's only available on Windows currently (more info [here](https://github.com/prusa3d/PrusaSlicer/issues/9066#issuecomment-1285090716)).
Perhaps you can find alternatives, there are a few model fixing services out there.

The model seems to be already broken on Printables since the issue is already detected when importing the file for the first time. So it's not caused by exporting the STL in PrusaSlicer, hence I'm closing this issue.

closed 2022-11-18 by DragonFinlay