prusa3d/PrusaSlicer #9156: I want to add fence mode

xiaogui8dangjia, 2022-11-14:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I often print models to turn over silicone materials，So it is necessary to circle around the model to pour silica gel
**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe how it would work**
Be as descriptive as your skill level allows you.

[image]
[image]
[image]
[image]
Increase the distance parameter between the model and the enclosure, and the thickness of the enclosure
This will be very convenient
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

DragonFinlay, 2022-11-17:
I'm sorry, I don't fully understand your request. What exactly does not work as expected for you, and what result would you like to see?

xiaogui8dangjia, 2022-11-18:
Automatically generate the wall of the model，
this is my model
[image]
and this is i want to 
[image]
make the wall of the model

DragonFinlay, 2022-11-18:
Hmm, so you basically need a copy of your outer walls with a certain distance from the inner model?
You can kinda get close to doing that with PrusaSlicer, but it's not quite possible in a clean way.
It's also too specific of a feature request to implement this in the near future. But we'll keep it open.

xiaogui8dangjia, 2022-11-20:
yes,I hope it will come true！

closed 2026-09-29 by prusaslicer-bot[bot]