prusa3d/PrusaSlicer #9165: Re-calculation of volumetric flow

MrBlackmidi, 2022-11-16:
**Is your feature request related to a problem? Please describe.**
As of right now, Prusaslicer calculates volumetric flow with mm^3 = speed x width x height instead of the more accurate stadium method explained here https://www.yogitech.eu/volumetric-flowrate-calculator-for-3d-printer/

**Describe the solution you'd like**
Making the volumetric clow calculation use the stadium method which is more accurate

**Describe how it would work**
Explained in the link above

rtyr, 2022-11-16:
> As of right now, Prusaslicer calculates volumetric flow with mm^3 = speed x width x height

What makes you think that? PrusaSlicer is not using such calculation.

MrBlackmidi, 2022-11-16:
Ive done the calculations and i have found that the values are very different, 3.021 mm^3 shown on prusaslicer, 3.39mm^3 on the site linked

rtyr, 2022-11-16:
We are calculating with rounded extrusions, we are not using that simple calculation.

closed 2022-11-16 by rtyr