prusa3d/PrusaSlicer #9173: Ender 5 plus - ASA filament

TymTucker, 2022-11-16:
### Description of the bug

Not sure if this is a temporary glitch or not.  I use Prusaslicer 2.5 at home and I just installed it at work.  I went to configure it to print ASA on the Ender 5 plus and it's telling me ASA is not compatible. It is on my slicer at home, and I currently have over 3000 hours printing nothing but ASA on the Ender 5 plus in a simple Creality enclosure.  Thanks. 

### Project file & How to reproduce

I'm not a coder. I don't know how any of that works.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

Windows 10

### Printer model

Creality Ender 5 plus

neophyl, 2022-11-17:
That's correct.  I've just run through the wizard and none of the Creality printers have a default ASA filament profile available to them.
Filaments and print profiles are part of vendor spaces and on top of that can also have other specific dependencies.  So for a profile to be available to a specific printer it must be in the same vendor space as the printer and also meet any dependency requirements.

I don't know where you got your ASA profile on your home machine.  Presumably its a detached one you got from somewhere.

Normally to get around this you would install a prusa printer profile, make sure the filament profile you want is installed and available to the prusa printer and then make a copy of that filament profile while you have the Prusa printer selected. 

Once you have a copy you then have to remove any dependencies, detach it from the system preset and then check for any other gotchas within the profile that could mess your printing up.  Things like having custom gcode in the filament that sets linear advance for example to something not suitable for your printer.  Its because of the custom gcode stored in most of the prusa specific filament definitions that they aren't compatible with other printers.

Dependencies and the detach from System button are in Filament>Dependencies which are all either Advanced or Expert level configuration items, so you need to be in expert mode ideally.

TymTucker, 2022-11-17:
Thanks.

For what it's worth, I have no idea what "vendor spaces" are, nor what a "detached one" is, nor most of the rest of what you said... I know the Prusaslicer 2.4 version I started with had Generic ASA available when I chose picked Ender 5 Plus (else I would not have been able to print it) so I was just trying to help the other technically illiterate out there.

Sent with [Proton Mail](https://proton.me/) secure email.

rtyr, 2022-11-18:
> I have no idea what "vendor spaces" are

It basically means that each vendor have it's own set of filament profiles. By default, you cannot enable Prusa3d profiles for Creality printers and vice versa.

"Generic PLA" profile for Prusa3d printers will be likely different than "Generic PLA" profile for Creality printers.

As of ASA. there is no ASA profile for Creality printers available at the moment, but it of course doesn't mean, that Creality printer cannot print such material. 3rd party profiles (such as Creality) are usually maintained by community members.

We are going to implement a collection of generic filament templates, which will be available for all printers across vendor spaces.

closed 2022-11-18 by rtyr