prusa3d/PrusaSlicer #9178: When starting prusa slicer i get this error "Internal Error: Calling AppConfig::save() from a worker thread"

paulrobin, 2022-11-18:
### Description of the bug

When starting prusa slicer 2.5.0+win64 I get warning and an error then a termination 

first dialog box = Unhandled unknown exception; terminating the application
[image]

second dialog box = "Internal Error: Calling AppConfig::save() from a worker thread"
[image]

Application exits 

happening on both 2.5.0 and 2.4.2

I have tried clearing the ini files, re-installing, removing all files from appdata/local and roaming aswell as the program files. even tried going back to 2.4.2 but getting the same error. 

Any other suggestions as to what could be going on here?

### Project file & How to reproduce

Unable to get to this as the application exists 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0 + 2.4.2

### Operating system

Windows 11

### Printer model

Voron 2.4

DragonFinlay, 2022-12-01:
Please check #7839, perhaps something there helps. Otherwise please give us as much info about your system (especially hardware) as possible.

DragonFinlay, 2022-12-16:
Author contacted us via e-mail, feedback:

It seems to have been related to the Nvidia control panel settings not being set to "Let the 3D application decide".

[image]

Closing as resolved.

closed 2022-12-16 by DragonFinlay

bubnikv, 2023-02-07:
Hopefully fixed with 84722876012ca310e29b291e10fef9d18ae26cea
Implementing a test whether the current thread is the main (UI) thread
and using it on AppConfig::save() to assert if save is called
from a worker thread.
The old assert was using thread names, which did not always work
on Windows.

MartynJ970, 2023-04-02:
having the same issue. Changing the Nvidia 3D aplication setting does not resolve the issue.

lukasmatena, 2023-04-02:
@MartynJ970 Having the same issue in which PrusaSlicer version? I would be very surprised to see this in 2.6.0-alpha6.