prusa3d/PrusaSlicer #9183: Purge Block Disappears with Small Parts with MMU2

captclearleft, 2022-11-19:
### Description of the bug

Purge Block Disappears with smaller parts that are Multi-Material.  Images are full size that slices with Purge Block, and slightly smaller where purge block disappears and part only prints single filament.
[image]
[image]
[image]
[image]

### Project file & How to reproduce

[Wood Paddle.zip](https://github.com/prusa3d/PrusaSlicer/files/10047864/Wood.Paddle.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

Windows 10

### Printer model

i3 MK3s MMU2s

captclearleft, 2022-11-19:
The images appear out of order.  Match the size of the Model to its sliced pair.

rtyr, 2022-11-19:
That's because that black part (extruder 3) gets too thin (something like 0.1mm), so it isn't printed at all. Since that part isn't printed, wipe tower is also not needed.

You have basically 3 main options:

closed 2022-11-19 by rtyr

captclearleft, 2022-11-19:
Thank You!!! It was modeled at .2 mm and I think you are right.  Thank You So Much