prusa3d/PrusaSlicer #9184: STEP Import Interpolation Resolution

kimkash, 2022-11-20:
### Description of the bug

I'm noticed that when importing STEP file to PrusaSlicer, some surfaces seems to be resolution dependent on the original data and prone to scaling problems. A bit vague, let me expand:

model of 'egg carton function' ( z=sin(x)+sin(y) ) being modeled in SOLIDWORKS 2017 using two sketches of sine wave and loft operation. Model being exported as STEP AP214.
Original export is very small because full sine wave interval (2pi=3.14) being interpreted as 3.14mm, so scaling is required.
When imported at x1 and being scaled x25 using PrusaSlicer built-in scaling modifiers (percentages), the result looks very grainy.
When scaled inside SOLIDWORKS to x5 being scaled x5 in PrusaSlicer the graininess looks better and when scaled x25 in SOLIDWORKS and no scaling at all at PrusaSlicer- results looks best.

I'm not sure it is a problem because I know that STEP stores geometric data as equations and not points/vertices/triangular faces but it seems like if there's internal conversion to STL when importing STEP then it should be handled better/differently. Scaling artifacts shouldn't be happening when using STEP.

Added 3 export files from SOLIDWORKS 2017, Project file with all 3 STEPs laid out and Screenshot angled to best show the differences.

### Project file & How to reproduce

[Part1.zip](https://github.com/prusa3d/PrusaSlicer/files/10049231/Part1.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

Win10 22H2 build 19045.2251

### Printer model

Prusa MK3S+ Stock

stepikovo, 2022-11-20:
Thanks for the report. STEP files are indeed getting triangulated in PrusaSlicer, we will check if it's possible to handle it differently.

stepikovo, 2022-11-23:
So after a discussion with the devs, it looks like we won't change how it's being handled.
The only way to do something about it is to reload the model after the scaling. Reloading the model will however undo fixes by the Netfabb, painting, modifiers and so on.

I'll leave the issue open for now.

Sembazuru, 2023-01-07:
Maybe a workaround to consider is to allow the user to choose a scaling factor when loading (or reloading) a step file (or other solid object format in the future) into PrusaSlicer. That way PrusaSlicer can tessellate at close to the intended actual print size.

Lukrative525, 2023-05-05:
Similar to @Sembazuru's suggestion, you could allow the user to specify a "step file import resolution" in Configuration > Preferences.

closed 2026-09-29 by prusaslicer-bot[bot]