prusa3d/PrusaSlicer #9185: "First layer height can't be greater than nozzle diameter" -- 0.5 mm layer height with a 0.8 mm

JoshLRR, 2022-11-20:
### Description of the bug

I'm trying to slice a part with a 0.5mm layer height and using an 0.8mm nozzle, but get the error in the title. Obviously, 0.5 is not greater than 0.8 so this error is incorrect. I have verified that the machine settings are correct.

### Project file & How to reproduce

1. Try to slice part using settings I already mentioned
2. Get cool error and can't export anything because the button is greyed out saying "invalid data"

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

windows 11

### Printer model

E3D ToolChanger. I guarantee this is useless information.

JoshLRR, 2022-11-20:
Had to work around this issue by changing the diameters of all my extruders despite them not being active in the print.

rtyr, 2022-12-13:
No response. Closing.

closed 2022-12-13 by rtyr