prusa3d/PrusaSlicer #9186: encore une perte des rayures

18pool49, 2022-11-20:
### Description of the bug

[image]
voilà ce que je demande des rayure toute les 60 couches 
ca devrais etre cela 
[image]
mais quand je fais le Gcode c'est loin d'etre ce que j'espère  
[image]

c'est la même pagaille que la dernière fois ???????

### Project file & How to reproduce

[ROUNDED TWISTED VASE bug rayure.zip](https://github.com/prusa3d/PrusaSlicer/files/10050217/ROUNDED.TWISTED.VASE.bug.rayure.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

w10

### Printer model

A10M geeetech

18pool49, 2022-11-21:
> Copie à : 18pool49 ***@***.***>;
>    Author ***@***.***>
>
> So, it's a collision between spiral vase and extruder changes? No seam -> no toolchange.
> Possible workaround: disable spiral vase, set one perimeter, no bottom, no top. Then add in a few bottom layers via a height modifier. (If you add bottom layers without height range it will try adding stuff to the walls). Downside: you will get a continuous seam which might leak. But toolchange seams are likely to leak anyway.

je ne suis pas d'accord avec votre méthode , la semaine dernière tout fonctionnait correctement en mode VASE et pour les outils, c'est 2 couleurs un buse donc pas de raccord
>> A10M Geeetech << 
il doit y avoir encore une fois un paramètre de Windows qui ne convient pas a Prusa suite a une mise a jour de W10 comme la dernière fois

18pool49, 2022-11-21:
je ne suis pas d'accord avec votre méthode , la semaine dernière tout fonctionnait correctement en mode VASE et pour les outils, c'est 2 couleurs un buse donc pas de raccord
>> A10M Geeetech << 
il doit y avoir encore une fois un paramètre de Windows qui ne convient pas a Prusa suite a une mise a jour de W10 comme la dernière fois 

I don't agree with your method, last week everything worked correctly in VASE mode and for the tools, it's 2 colors one nozzle so no connection

>>A10M Geeetech<<

there must once again be a Windows setting that is not suitable for Prusa following a W10 update like last time

kubispe1, 2022-11-24:
Hi, I tested your issue and Iam able to reproduce it. We are trying to find solution. Sorry, but at the moment issue is not fixed. I link your new examples and notes to your previous one and close it. Thanks for understanding
#9095

closed 2022-11-24 by kubispe1