prusa3d/PrusaSlicer #9188: Crash on large MMU painting project

fract0id, 2022-11-21:
### Description of the bug

Prusaslicer crashes when I try to slice a large (fills a 350*350 Voron bed) and complex (>1 mio triangles) model with MMU painting and some height-based color changes. Running with log enabled shows that it crashes just after printing "MMU segmentation - merging segmented layers in parallel - end".

[image]

It doesn't crash:
- with just a little bit of MMU painting
- without MMU painting and only color based on height
- resizing to 50% in XY, same Z, same MMU painting
- on other relatively large, complex models with MMU painting

It does crash
- in PS 2.5 and 2.6 alpha built from github a few days ago
- in Superslicer, with a separate project of the same model, all settings and MMU painting done separately
- resized to 90%
- with all kinds of different settings that I could think of
- on both my old and my new computer, so its not hardware, OS or lack of memory

I tried running the 2.6 alpha I built in debug, but it was too slow (usually, its ~30s to crash, it ran > 30min without getting close to the MMU segmentation step) and I canceled before a crash.

The project file is 35 MB zipped, I uploaded it here: https://www.swisstransfer.com/d/0e633087-4785-4c51-badb-5a6cfd9e6a0f

Anything to try or any possible workarounds are greatly appreciated :)

### Project file & How to reproduce

Open file, hit slice. Wait a few minutes if you computer is slow, on my 7950x it takes 30s.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0 and 2.6 alpha

### Operating system

Win 10 and 11

### Printer model

Voron Trident 350 with ERCF (MMU)

kubispe1, 2022-11-24:
Hi, I tried on Win and can reproduce the issue.

- issue is in all versions 2.6.0, 2.5.0, 2.4.2
- MMU Painting and height range modifiers works separately fine, but together it crashes.
- it is single extruder MMU

We will investigate it more. Let's see thank you @fract0id

fract0id, 2022-12-06:
Hi. Is there any news? Any possible workaround? Anything I can do to help?

fract0id, 2023-01-05:
I hunted down the trigger conditions and was able to make a very simple file (attached below) to reproduce the crash. It seems to happen when the layer-based color change occurs at the same time as a paiting-based color change. 

With this file I was able to run debug and see exactly where the program crashes, but so far I could not figure out where the actual origin of the issue is.

The debug message says "can't dereference out of range vector iterator" and the problem seems to occur around line 585 of PrintObjectSlice.cpp. 

After the crash, the debugger shows me invalid values for it_painted_region->extruder_id and region_id, but I am not sure if those actually occur during the execution.

Could anyone more familiar with the code or at least with the VS debugger help me figure this out?

[PS MMU painting crash trigger.zip](https://github.com/prusa3d/PrusaSlicer/files/10351823/PS.MMU.painting.crash.trigger.zip)

closed 2026-09-29 by prusaslicer-bot[bot]