prusa3d/PrusaSlicer #9189: perte de sélection de couleur

18pool49, 2022-11-21:
### Description of the bug

quand je demande une sélection de couleurs >> ca ne marche pas a chaque fois 
et quand ca marche  (heureux) c'est bien représenté en couleur a l'écran 
[image]
mais si je découpe  je perds mes sélection a l'écran 
[image]
et je n'ai pas de rayure de couleurs sur ma pièce donc pas de changement d'extrudeur 

### Project file & How to reproduce

[ROUNDED TWISTED VASE bug bis rayure.zip](https://github.com/prusa3d/PrusaSlicer/files/10057616/ROUNDED.TWISTED.VASE.bug.bis.rayure.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Édition	Windows 10 Famille Version	22H2 Installé le	‎sam. ‎25/‎sept./‎2021 Build du système d’exploitation	19045.2311 Expérience	Windows Feature Experience Pack 120.2212.4190.0

### Printer model

Geeetech A10M

18pool49, 2022-11-21:
une fois demandé la découpe soit disant réalisée je n'ai pas le Gcode puisque  dans la prise d'écran (Pièce jaune) il y a encore >>
DÉcouper maintenant << en bas a droite  ? ? ? ? ?

18pool49, 2022-11-21:
et en plus la pièce est jaune a l'écran alors que dans le cartouche de droite sur la ligne de pièce la couleur SPÉCIFIER est rouge 

il y a tout POUR PLAIRE 

et rien qui fonctionne

kubispe1, 2022-11-24:
Hi, we are trying to discover problem. Sorry, but at the moment issue is not solved. I link this issue to your previous one and close it. Thanks for understanding
#9095

closed 2022-11-24 by kubispe1