prusa3d/PrusaSlicer #9191: Wipe object don't appear to reduce size of wipe tower

shadow7412, 2022-11-21:
### Description of the bug

I was trying to set up my slicer to purge into objects, and while the setting appears to be working (in the sense that those objects now contain many colours) the wipe tower dimensions didn't change at all.
[image]

I took this to extremes to confirm my suspicion.
[image]

If there is sufficient "wipe" occuring in the wip object, then a wipe tower (particularly for that layer) should be redundant - so keeping it there only wastes filament.

### Project file & How to reproduce

[snail.zip](https://github.com/prusa3d/PrusaSlicer/files/10061025/snail.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0

### Operating system

Manjaro

### Printer model

Geeetech A10M

bombela, 2022-12-14:
Double check this setting:
[image]

shadow7412, 2022-12-21:
It was set to 15mm3. Changing it to 0 did certainly help.

Looking inside the tower, I noticed that it's actually quite hollow. So its height is maintained for some reason... but it's certainly using less filament now. Thanks for the tip.

bombela, 2022-12-21:
You can also use the experimental sparse layer feature of the wipe tower.

lukasmatena, 2022-12-22:
@shadow7412 The wipe tower is hollow because it cannot widen at the top, where there is nothing else to wipe into. If you cut your snail in half, the hollow space will go away.

Can we close the issue?

shadow7412, 2022-12-22:
I don't fully understand what you mean... the wipe object is the large circular shape on the right. The snail is the "main" object.

But you can probably close this (unless you feel it'd be helpful to elaborate on the mechanics of wipe objects in the documentation).

lukasmatena, 2022-12-22:
You wrote:

> Looking inside the tower, I noticed that it's actually quite hollow.

From this I assumed that you are talking about the scenario where the big object is set as unprintable, because in the other case the wipe tower is not hollow. And I was just explaining why the wipe tower is hollow.

closed 2022-12-22 by lukasmatena

kurtgluck, 2025-04-10:
This seems to be a bug to me.   I have tried Wipe into object AND/OR wipe into infill with 4 objects with those off and 2 objects with them on.  In all cases the 2 objects that you would expect to be used for wipeing slice identically to the normal ones UNLESS the wipe tower is turned on.  BUT if the wipe tower is turned on that eliminates the reason I would want to use the feature.   I think there is a problem here whereby the wipe tower flag is turning off ALL wipeing