prusa3d/PrusaSlicer #9192: No Access to Used Filament statistic where it's useful / Custom G-Code Substitutions not parsing comments

codefaux, 2022-11-21:
### Description of the bug

The title sucks, but I have a problem I can fix one of two ways.

I need to place a GCode command at the end of my GCode file, informing my printer how much filament I've just used.

My initial intent was to use `[filament_used]` in **Printer Settings** -> **Custom G-Code** -> **End G-Code**
[image]
[image]

My second attempt was to use **Print Settings** -> **Output options** -> **Other** -> **G-Code Substitutions**
[image]
[image]
(PS: I know [used_filament] won't/shouldn't work here, it was a desperation attempt.)

I see that `used_filament` IS a placeholder, but ONLY in the output filename - making this (and most other placeholders) accessible in Printer Settings -> Custom G-Code seems entirely reasonable.

If this is unaccepable, can **Print Settings** -> **Output options** -> **Other** -> **G-Code Substitutions** be modified to parse the **ENTIRE** gcode file, specifically including comments? There is a lot of information provided in gcode comments, which would be useful for either LCD display lines or info tracking.

### Project file & How to reproduce

Project not applicable, core functionality

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-win64

### Operating system

Windows 10

### Printer model

COMPLETELY custom build, Klipper + dual MCU

bubnikv, 2022-11-24:
Currently custom G-code blocks do not support placeholders for print
statistics, because the placeholders are processed during G-code export
while print statistics values are known first after the G-code is exported.
It is technically doable to postpone processing of print statistics values
in placehodlers, but it requires some additional work and it will slow down
G-code export.

Placeholders are not supported in G-code substitutions. Even if they would,
the same limitation applies as above: The G-code substitutions are
processed during G-code export.

codefaux, 2022-11-24:
> technically doable [...] but
Understood and accepted.

> Placeholders are not supported in G-code substitutions
Not what I was asking for, I specifically did not expect that to work. As I mentioned in my Issue, it was a last-ditch effort.

The other lines of G-Code substitution in the screenshot also did not work, none of them required placeholders -- they required processing comments.

I understand that you* (abstracting 'you' as 'the devs' because it's easier) don't want to process comments because users might/consistently do break things -- if "Expert" on the settings window isn't enough, I have a suggestion. Optionally, a checkbox alongside the others per-substitution to `"Also process comments (WARNING)"` with tooltip text which boils down to `"This can break in unexpected ways, especially with badly formatted regexes"` would cover the requirements for sufficient risk disclosure while still adding functionality beyond just my own.

This way, users who desire comment processing get their way -- at the clearly acknowledged risk of breakage -- and normal users (who probably aren't using this anyway) don't need to fuss or change expectations. It seems an acceptable compromise, and I do genuinely mean to target more than my own interests with propositions, because "do this for ME" is an absolutely unreasonable ask.

realtw, 2022-11-26:
offtopic as in just a suggestion, but if the placeholder works, save into filename and maybe use another script to process the filename into a gcode append to file?

codefaux, 2022-12-03:
> offtopic as in just a suggestion, but if the placeholder works, save into filename and maybe use another script to process the filename into a gcode append to file?

- Fair enough, that is the in-place available solution and it would function.
- If I can process the file with a script, I don't need the information saved into the filename -- the information is already in the file, as a comment
- I don't know how to write that script
- It seems redundant to write a regex script to process a file after the regex service within the slicer processes the file.
- My printer's profile is saved on a network path, and used w/ multiple machine/OS combinations. It is (for me, in my situation) non-trivial to set up a "universal" processing script which executes with zero changes on multiple machine/OS combinations (Windows, Linux, etc)
- I'm also trying to reduce dependencies, software, and extra steps -- not increase it.
- It would be much easier to use the already-in-place Regex substitution service that's included with the slicer, if it's possible to get it to optionally process comments for Expert users.

Agent-Cain, 2023-11-24:
> Currently custom G-code blocks do not support placeholders for print statistics, because the placeholders are processed during G-code export while print statistics values are known first after the [...]

I just used the new g-code editor tool of PS 2.7. 
It gives the possibility to set the variable used_filament under  Print statistics, but it is still not usable ?
That seems a bit odd.

closed 2026-09-29 by prusaslicer-bot[bot]