prusa3d/PrusaSlicer #9194: Prusa 2.5.0 does not allow to save settings profile

Algtronic, 2022-11-22:
### Description of the bug

I am not allowed to set imported settings as primary settings. However, I can modify and export these settings then import them again. I think saving must be allowed to imported configuration.

### Project file & How to reproduce

Import settings file then try to save it as your primary

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

win64

### Printer model

ender 3

rtyr, 2022-11-25:
You have to change the name while saving it.

Can we close the issue?

Algtronic, 2022-11-25:
It is not intuitive enough because settings name should not relate to filename. I think you should fix this. But for now it's closed, yes

closed 2022-11-25 by Algtronic