prusa3d/PrusaSlicer #9196: Can we eliminate tangent lines to holes when printing vertical wall?

laptop-laptop, 2022-11-23:
**Is your feature request related to a problem? Please describe.**

When printing a vertical wall-like structure with some holes, there are tangent lines at the top-most and bottom-most extremes of the holes. Such lines run across the entire structure, so it is easy to tell the layers. They seem to be even more pronounced when Variable Layer Height is used.

What are they? Is there any option to make them disappear or less obvious?

laptop-laptop, 2022-11-24:
This sort of lines also shows up when there is a change in the shape. For example, print a vertical wall of 10mm thick up to height 50mm. Then continue to print a vertical wall of, say 7mm thick.

DragonFinlay, 2022-12-08:
Can you please send pictures and an example 3MF to visualize what you mean?

laptop-laptop, 2022-12-10:
Without variable layering, although visualization looks fine. Sometimes I can see horizontal lines on the actual prints.
With variable layering, these horizontal lines became more visible. The more times I press the Smooth button, the more
obvious the lines show up on the actual prints. Can we eliminate them or make them as invisible as possible?

Without variable layering:

[url=https://imgur.com/1fPddNE][img]http://i.imgur.com/1fPddNE.jpg[/img][/url]

With variable layering (Adaptive 1, Smooth 1 and pressed Smooth once):

[url=https://imgur.com/TG7w5LE][img]http://i.imgur.com/TG7w5LE.jpg[/img][/url]

With variable layering (Adaptive 1, Smooth 1 and pressed 10 times). 

[url=https://imgur.com/lmbi45k][img]http://i.imgur.com/lmbi45k.jpg[/img][/url]

laptop-laptop, 2022-12-10:
I forgot to mention that such strange lines show up not only on completely flat vertical walls but also on curved surfaces.

laptop-laptop, 2022-12-12:
Here is an example of ugly lines on curved surface. I have a rectangular box and a hole on one side. After using variable layering, you see that there are unexpected lines surrounding the cylinder at the levels where the top and bottom of the rectangular and hole are located. Adaptive and Smooth buttons were pressed once. If I press the Smooth buttom more, more ugly lines show up. Is it possible to make them not showing up in the real print?

[image]

DragonFinlay, 2022-12-16:
Okay, I see now what you mean. This is a pretty common issue with FDM printing, closely related to the Benchy Hull Line.
Do you have photos of finished prints with the line visible so we can see the comparison?

closed 2026-09-29 by prusaslicer-bot[bot]