prusa3d/PrusaSlicer #9197: No ironing on 100% infill surfaces

KlausHans, 2022-11-23:
### Description of the bug

Settings:
- 0 top an 0 bottom layers
- 100% infill (aligned) rectilinear
- ironing on top most surface or all top surfaces enabled

Expectation:
- the top most surfaces gets ironed

Actual result:
- no ironing at all

Additional info:
- i want to create a very clear print with transparent filament. That's why i can't use top or bottom surfaces but want ironing none the less

### Project file & How to reproduce

[OledCoverClearPETG.zip](https://github.com/prusa3d/PrusaSlicer/files/10079188/OledCoverClearPETG.zip)
[image]

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.5.0

### Operating system

Ubuntu 22.10

### Printer model

Voron 2.4

DragonFinlay, 2022-12-16:
The issue here is that PrusaSlicer doesn't recognize these surfaces as "top" or "bottom". Even though it's functionally the same, PrusaSlicer sees it as simply perimeters with infill.
I've done glass printing myself and I encountered the same issue, but this is quite a specific edge case. For now, you can only use ironing on actual solid layers.

KlausHans, 2022-12-16:
@NGC869 The model is printed in 4 layers, not two.
If i use 1 "real" top layer i get this on the second layer:
[image]
As you can see ironing is not usable for "glas" printing.

closed 2026-09-29 by prusaslicer-bot[bot]