prusa3d/PrusaSlicer #9203: Paint on brims

lucasmoes, 2022-11-25:
**Is your feature request related to a problem? Please describe.**
When I am printing larger objects I get warping problems but, one of the solutions is that you can add a brim but for some parts of the print a brim is not wished for when is comes to parts of the print where there is more accuracy needed.

**Describe the solution you'd like**
What would be nice is that you can paint brims your self where you want a brim te appear.

**Describe how it would work**
Similar to the paint on support, but paint anywhere on the object that is touching the build plate. when painting on the brim, the object will be colored maybe about 3mm from the build plate to show where the brim would appear in a different color then the support painting.

**Describe alternatives you've considered**
cura slicer had this feature where you have corner brims but, the way I described it, would feel better.

**Additional context**
-

rtyr, 2022-11-25:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/8283.

closed 2022-11-25 by rtyr

Snuff1eupagus, 2024-04-26:
[lucasmoes](https://github.com/lucasmoes) this may help you. I found this very enlightening.

https://www.youtube.com/watch?v=dhsbMmosbLU&t=4s

Good Luck