prusa3d/PrusaSlicer #9205: Report of license infringement on a unpublished fork and distribution

sn4k3, 2022-11-25:
### Description of the bug

- [GKone Slicer](https://www.uniformation3d.com/pages/gkone)
- [GKtwo Slicer](https://www.uniformation3d.com/pages/gktwo)

They are clearly modified forks of PrusaSlicer which uses AGPL and code is not disclosed as I can find.
Named as `Slicer` and used for two mSLA printers of uniformation.

GKone uses .sl1 variant (jxs) and GKtwo uses .ctb disguised of .jxs

I could not find any source of the fork nor a link to it, so I guess the license is under infringement.

[image]

### Project file & How to reproduce

Download gkone or gktwo slicer, run and inspect.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.0

### Operating system

Windows 11

### Printer model

GKone and GKtwo

bubnikv, 2022-11-28:
Thanks for heads up.
We may complain here :-)
https://www.uniformation3d.com/pages/intellectual-property-rights

clytle374, 2024-11-13:
I was looking to buy a GKtwo.  But saw this.  
Has this been resolved?

closed 2026-09-29 by prusaslicer-bot[bot]