prusa3d/PrusaSlicer #9206: Prusa Slicer 2.5.0 will not open STL files

CyberBiker, 2022-11-25:
### Description of the bug

unable to open stl files with prusa slicer 2.5.0

### Project file & How to reproduce

Screenshot 2022-11-25 at 5.30.30 PM

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0+MacOS-arm64

### Operating system

Mac OS 13.0.1

### Printer model

Original Prusa i3 mk3

CyberBiker, 2022-11-25:
[image]

kubispe1, 2022-11-29:
Hi, can you provide your .stl ? Let's check it out... :D

bubnikv, 2022-11-29:
Looks like you are opening a project file, not importing an object.

bradleymadaris, 2022-12-01:
@CyberBiker, it appears you're attempting to import an .STL file by selecting Open Project instead of Import > Import STL/3MF/STEP/OBJ/AMF..., see below.

This is me attempting to import an .STL by selecting Open Project.
[image]

This is me attempting to import an .STL by selecting Import > Import STL/3MF/STEP/OBJ/AMF...
[image]

Can you confirm if this is the same for you?

closed 2022-12-18 by stepikovo