prusa3d/PrusaSlicer #9209: Can't save directly to SD card

clevijoki, 2022-11-26:
### Description of the bug

Starting sometime in the past couple weeks, I'm unable to save directly to the SD card. This was working before with version 2.5.0.

It pops up a dialog box saying "You can't save to D: Try another location"

The SD card is not locked, and I can save to the desktop and copy the files manually after.

I suspect it was a recent windows update that may have broken it, as I recently installed KB5019980. 

### Project file & How to reproduce

[locker-level-high.zip](https://github.com/prusa3d/PrusaSlicer/files/10095469/locker-level-high.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0+win64

### Operating system

Windows 11, 22H2, latest patches circa Nov '22

### Printer model

Prusa MK3S+

kubispe1, 2022-11-29:
Hi, I simply tried saving your .3mf to SD card and everything works fine.
Have you also tried formatting the SD card or using a different one?

clevijoki, 2022-11-30:
Well I am not sure what happened. I grabbed the latest source so I could try debugging it myself, and it didn't happen on 2.6 alpha.

Then I tried to repro on 2.5 and again it worked. 👻

closed 2022-11-30 by clevijoki