prusa3d/PrusaSlicer #9210: V2.5 Plate Markings Visual

antrux1, 2022-11-26:
Just updated to 2.5 from an older version and while I do think the new visual for the on screen photo realistic print bed is cool when all the lines change patterns while turning the bed onscreen the lines are distracting. For me it's bearable but I much prefer the static lines of the previous version. Possibly an option to keep the previous visual style is needed.

[image]

JohnOCFII, 2022-11-26:
What computer/operating system/and video card are you using?  PrusaSlicer 2.5.0 on my 2021 MacBook Pro does not look like that to me.
[image]
[image]

antrux1, 2022-11-27:
Oh wow, ok. It looks cool so I thought it was some new visual function but
it hurts when lining up multiple objects on the bed.
FYI, I did also right before updating to v2.5 temporarily adjust the bed
size in order to get a part to print. I adjusted the 250mm x 210mm to 250mm
x 215mm and once finished set the values back.

*My Info:*
Processor Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz   3.70 GHz
Installed RAM 16.0 GB
Product ID 00330-80000-00000-AA789
System type 64-bit operating system, x64-based processor
Pen and touch No pen or touch input is available for this display
Edition Windows 10 Pro
Version 21H2
Installed on ‎3/‎8/‎2021
OS build 19044.2251
Experience Windows Feature Experience Pack 120.2212.4180.0
Three monitors each at 1920x1080, No scaling.
Radeon RX570 card Driver v31.0.12029.7019 *(Just updated today)*

PrusaSlicer
Version:   2.5.0+win64
Build:     PrusaSlicer-2.5.0+win64-202209060714

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19044), 64-bit edition
Total RAM size [MB]: 17,141MB
OpenGL installation
GL version:   4.6.0 Compatibility Profile Context 22.11.1.221110
Vendor:       ATI Technologies Inc.
Renderer:     Radeon RX 570 Series
GLSL version: 4.60
<details>
<summary>Installed extensions:</summary>
GL_AMD_blend_minmax_factor
GL_AMD_bus_addressable_memory
GL_AMD_depth_clamp_separate
GL_AMD_gpu_shader_half_float
GL_AMD_gpu_shader_int16
GL_AMD_gpu_shader_int64
GL_AMD_pinned_memory
GL_AMD_sample_positions
GL_AMD_shader_trinary_minmax
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compute_shader
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
[... 196 lines cut ...]
GL_EXT_texture_storage
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array
GL_KHR_blend_equation_advanced
GL_KHR_debug
GL_KHR_no_error
GL_KHR_parallel_shader_compile
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_NV_blend_square
GL_NV_copy_image
GL_NV_depth_buffer_float
GL_NV_float_buffer
GL_NV_primitive_restart
GL_NV_shader_atomic_int64
GL_NV_texture_barrier
GL_OES_EGL_image
GL_OES_draw_texture
GL_OVR_multiview
GL_OVR_multiview2
GL_OVR_multiview_multisampled_render_to_texture
GL_S3_s3tc
WGL_EXT_swap_control
</details>

bubnikv, 2022-11-28:
This issue is caused by a faulty OpenGL driver messing up texture mip
mapping.
If it is a new issue for you, you may try to upgrade or downgrade your
graphics driver.

antrux1, 2022-11-28:
After your first reply I updated all related graphics software, maybe a
year outdated, and rebooted to check. After the issue was still there I
copied info and replied. Any info on a specific part, API call, dll, or etc
because downgrading versions will be going back maybe a couple years back
now.

Joe

lukasmatena, 2023-03-27:
Do you still have this issue with 2.6.0-alpha5 ? Thanks.

antrux1, 2023-07-29:
I was unable to compile that version for windows on my own however as of
v2.6.0 yes the issue is now gone.
Thank you very much!

Joe

closed 2023-07-29 by lukasmatena