prusa3d/PrusaSlicer #9212: soluble support setting preventing wipe tower even though i just want my supports to be a different color

TheJoyofPrinting, 2022-11-26:
### Description of the bug

I want to have my support be a different color not soluble. How ever prusa slicer prevents the wipe tower from being used if you set the supports to any extruder but 0 as it thinks im using soluble supports. I should be able to tell the slicer if its soluble or not and get the wipe tower based on that. For example im printing a zelda shield that is really high off the build plate. I dont want to waste my good silver filament for the support and instead use a color I never use/dont like. 
[image]
[image]

### Project file & How to reproduce

[hylian_shield-body.zip](https://github.com/prusa3d/PrusaSlicer/files/10096818/hylian_shield-body.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

windows 10

### Printer model

prusa i3 mk3s

rtyr, 2022-11-26:
This is not currently supported as described in the tooltip. Support material uses variable layer height, independent from main object, only with soluble supports the layers are synchronized with the object. 

>  dont want to waste my good silver filament for the support

This will not help much, print time will be much higher and the "good" filament will be also used during toolchanges.

I am closing this issue, expected behavior described in the tooltip.

closed 2022-11-26 by rtyr

TheJoyofPrinting, 2022-11-26:
wow so no help and just closing it because you think its not needed and there is no use case? you cannot be serious? why am i forced to use the settings the way you want me to? why can i not use this without it thinking its soluble? terrible but ok...

rtyr, 2022-11-26:
It is known and intended behavior described in tooltip. Also a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3981.