prusa3d/PrusaSlicer #9213: Mk2.5 can't be setup as PrusaLink

mstandif, 2022-11-26:
### Description of the bug

Prusa Mk2.5 settings menu lacks the option to setup the printer for PrusaLink gcode upload.

### Project file & How to reproduce

In the Configuration Wizard, have the Prusa Mk2.5 0.4mm system preset added
Select the Prusa Mk2.5 preset
Go to the "Printer Settings" tab
Click the gear icon to the right of the preset drop-down to add a physical printer
Under "Host Type" there's no option for PrusaLink 

I've included images showing the Host Type menu for both the Mk2.5, and the Mk3s

[Untitled.zip](https://github.com/prusa3d/PrusaSlicer/files/10097206/Untitled.zip)
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-win64

### Operating system

Windows 10

### Printer model

Prusa Mk2.5

kubispe1, 2022-12-08:
Thank you for issue. Should be corrected in next release 2.6.0

kubispe1, 2023-01-12:
It is merged to the master of 2.6.0. Thanks
[image]

closed 2023-01-12 by kubispe1