prusa3d/PrusaSlicer #9214: Funny slicing angle

FrHePo, 2022-11-27:
### Description of the bug

Prusa slicer uses a slight angle while bridging a gap. The optimal bridging of this gap would be without angle. There is no part contour that could cause this non optimal angle.

### Project file & How to reproduce

[Why this angle.zip](https://github.com/prusa3d/PrusaSlicer/files/10097396/Why.this.angle.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.5.0+win64

### Operating system

Win10

### Printer model

Prusa MK3

DragonFinlay, 2022-12-13:
It seems to be related to the Voronoi diagram generator. If you rotate the part by any amount (e.g. 0.01°), the infill will be generated with the correct angle.

[image]

hejllukas, 2024-04-16:
It seems to be already fixed in PrusaSlicer 2.7.4. So, I'm closing this issue.

closed 2024-04-16 by hejllukas