prusa3d/PrusaSlicer #9217: Make low-height wipe tower if possible

vovodroid, 2022-11-27:
Currently wipe tower has full height of object, that could be huge waste of material and time. Assuming we have single tool change near object top:

[image]

(besides some strange artifact in the air).

Why not descend extruder and wipe at bed level before actual color change (if place allows, of course)?

rtyr, 2022-11-27:
Already implemented. Multiple extruders -> No sparse layers.

closed 2022-11-27 by rtyr

vovodroid, 2022-11-30:
> Multiple extruders -> No sparse layers.

Thanks! Nevertheless what about extruder dimensions and tower location check as for sequential printing?

[image]

Won't extruder collide to the object?

rtyr, 2022-11-30:
Yes it will. It is currently up to the user to use safe wipe tower position. It is mentioned in the tooltip.

vovodroid, 2024-01-24:
@rtyr 

> See FAQ for "strange artefact" above wipe tower (it is completely normal/expected).

Where can I find this FAQ?