prusa3d/PrusaSlicer #9218: Add a "Reveal in File Explorer" button to per-model context menu

DeflateAwning, 2022-11-27:
**Is your feature request related to a problem? Please describe.**
Sometimes I want to make a quick edit to a file (using a CAD program) for a model that I've already added to the build plate. Sometimes, I've already closed Windows Explorer/Finder/File Explorer window, and have to navigate to the file again.

**Describe the solution you'd like**
I want a right click menu item added that works like this:

1. Right click on a model on the build plate, or right click on a model in the item list in the right pane.
2. Click the new menu item "Reveal in File Explorer"
3. A Windows Explorer/Finder/File Explorer window would pop up, with the specific STL file selected/highlighted. 

**Describe alternatives you've considered**
Without this solution, you have to manually navagate through your filesystem to the model.

**Additional context**
This sort of feature is super common in the following locations:
* Right clicking on a file search result when searching files in the filesystem.
* In most web browsers, when you've downloaded a file, there's a similar mechanism to show that file in a file explorer.

It makes perfect sense to me that it should be added here.

neophyl, 2022-12-13:
Are you not using the right click on the existing object on the plater and reload from disk option ?  Normal workflow is to output your stl/file.  Load it to the plater and do whatever needed, decide to edit the model in CAD, export the updated version and then switch back to PS and use the reload from disk option.

DeflateAwning, 2022-12-13:
I understand that the "Reload from disk" option exists.  I want an additional button added that's called "Reveal in file explorer" (alongside the "Reload from disk" option).

I understand the "normal workflow". Here's my workflow sometimes:
1. Design something using OpenSCAD (or any other software) in a folder that's very well organized (deep in Google Drive > CAD > Categories > Projects > 2022 > Handles > Handle Design 3 > Modified from Thingiverse > MyHandle.scad). 
2. Export to STL.
3. **Close OpenSCAD**.
4. Pat myself on the back, and say "wow that'll work perfectly". 
5. Drag the file into Prusaslicer. 
6. **Close the folder in File Explorer**, because it's literally perfect.
7. Slice it.
8. Realize I screwed up wall thickness, and that I have to modify the file and re-export it.
9. In Windows Explorer, navigate 10 layers deep in my filesystem to get to the OpenSCAD file I want.

In my workflow, I'd be able to skip the last step of navigating through my filesystem if there was a "Reveal in file explorer" button. I wouldn't use it all the time, but it'd be helpful for that once-in-a-while when I accidentally close the file and close Windows Explorer.

neophyl, 2022-12-13:
So Openscad doesnt have any recent file list I take it ?  
Does your OS have a recent folder list ?  I know windows does as part of the Quick access list that shows the most recently used folders and files.  Just trying to think of alternatives.

DeflateAwning, 2022-12-13:
I generally duplicate the file and add a version number to it (i.e., Handle v1.scad, Handle v2.scad, etc.), which can't be done in the OpenSCAD recent files menu.

Quick Access is not a list of recent folders; it's a list of recent folders, weighted by access frequency. If I simply edit a file in there, it doesn't show in the Quick Access pane. 

Another use case - Sometimes I'll have a file open in Prusaslicer for days while it's printing, or after it's printed but before I get a chance to look at it. By that point, it's been days since I've accessed the file. In such a case, it would be helpful to get back to the file from within Prusaslicer.

neophyl, 2022-12-13:
In windows 10 (not updated to 11 as yet) if you click on the top level quick access entry it brings up a curated list of recent folder yes, but it also displays the 20 most recent files used too.

DeflateAwning, 2026-09-02:
This issue would probably be best to convert to a discussion (as it's a Feature Request). It's not clear to me how to do that though.