prusa3d/PrusaSlicer #9219: Detect thin walls - cannot check it, keeps disabled

Eldenroot, 2022-11-27:
### Description of the bug

Go to your Prusa Slicer settings and try to enable the "detect thin walls" - cannot add a check mark!
[image]

### Project file & How to reproduce

Go to your Prusa Slicer settings and try to enable the "detect thin walls" - cannot add a check mark!

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Windows 11 64bit

### Printer model

Prusa Mini+

rtyr, 2022-11-28:
This option is disabled with arachne slicing engine. It is available only with classic slicing engine.

closed 2022-11-28 by rtyr

Eldenroot, 2022-11-28:
Ok, but is should br greyed out or add some notice.

Now it is confusing!

rtyr, 2022-11-28:
It is normally greyed out, but some users see it non-greyed. It is already reported here https://github.com/prusa3d/PrusaSlicer/issues/8877