prusa3d/PrusaSlicer #9227: Remove blocking of defining the number of perimeters for the base section of vase mode print

firepebble, 2022-11-30:
**Description**
It would make vase much better to allow users to specify the number of perimeters for solid layers of a print. Version 2.5.0 will block users from specifying the number of perimeters when trying to: add a modifier shape or a part, and then adding a layers and perimeters setting to the shape and then specifying the number of perimeters.

**Describe the solution you'd like**
Allow the specification, through modifiers or parts, to the number of perimeters allowed in vase mode.

**Describe how it would work**
Add modifier, specify the number of perimeters - slicer would then only apply this to solid layers and warn users not being applied to single wall sections.

curieos, 2023-02-12:
In the same vein, I would really like the ability to modify the base of the print in spiral vase. Basically, treat the bottom portion as a normal print, then switch to spiral vase after a certain height. It would make printing [this Gridfinity component](https://www.printables.com/model/324102-vase-mode-parametric-gridfinity) and similar things really slick. Currently the solution is to either glue them post print, or splice G-code together, neither is very convenient.

I can currently set the bottom solid layers to a really high amount to achieve sorta what I want, though it's printing the base solid, which wastes time and material.

firepebble, 2023-02-13:
That would be really cool. As a bit of a hack, I add cut-outs to the solid layers in the model to do it manually. Quite annoying if you want to change print layer heights or the thickness of the base.

closed 2026-09-29 by prusaslicer-bot[bot]