prusa3d/PrusaSlicer #9229: Can't compile MASTER branch

yangminglong, 2022-11-30:
> D:\Dev\prusaSlicer\src\PrusaSlicer\build>cmake .. -G "Visual Studio 16 2019" -DCMAKE_PREFIX_PATH="D:\Dev\prusaSlicer\src\PrusaSlicer-deps\usr\local"
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19044.
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: ON
-- SLIC3R_DESKTOP_INTEGRATION: ON
-- SLIC3R_ENC_CHECK: ON
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: ON
-- SLIC3R_UBSAN: OFF
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: ON
-- CMAKE_PREFIX_PATH: D:\Dev\prusaSlicer\src\PrusaSlicer-deps\usr\local (from cache or command line)
/Include//winrt/windows.graphics.printing3d.h was not found
STL fixing by the Netfabb service will not be compiled
Building without Win10 Netfabb STL fixing service support
-- Found Boost: D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/Boost-1.78.0/BoostConfig.cmake (found suitable version "1.78.0", minimum required is "1.66.0") found components: system filesystem thread log locale regex chrono atomic date_time iostreams
-- Boost::boost exists
-- TBB found in D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/TBB
-- Found ZLIB: optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/zlib.lib;debug;C:/Program Files/zlib/lib/zlibd.lib (found suitable version "1.2.11", minimum required is "1")
-- CURL found in D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/CURL
-- EXPAT found in D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/expat-2.4.3
-- Found ZLIB: optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/zlib.lib;debug;C:/Program Files/zlib/lib/zlibd.lib (found version "1.2.11")
-- GLEW found in D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/glew
-- try find NLopt in NLopt_DIR:''
-- Found NLopt in 'D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib'.
-- Using NLopt include directory 'D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/include'.
-- Using NLopt library 'D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/nlopt.lib'.
CMake Warning at cmake/modules/OpenVDBUtils.cmake:129 (message):
  vdb_print not found! OPENVDB_PRINT-NOTFOUND
Call Stack (most recent call first):
  cmake/modules/FindOpenVDB.cmake:317 (OPENVDB_ABI_VERSION_FROM_PRINT)
  CMakeLists.txt:555 (find_package)

>CMake Warning at cmake/modules/FindOpenVDB.cmake:324 (message):
  Unable to determine OpenVDB ABI version from OpenVDB installation.  The
  library major version "8" will be inferred.  If this is not correct, use
  add_definitions(-DOPENVDB_ABI_VERSION_NUMBER=N)
Call Stack (most recent call first):
  CMakeLists.txt:555 (find_package)

>-- TBB found in D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/TBB
-- Found Boost: D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/Boost-1.78.0/BoostConfig.cmake (found version "1.78.0") found components: iostreams system
-- OpenVDB libraries: D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/libopenvdb.lib;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/libopenvdbd.lib
-- IGL NOT found, using bundled version...
CMake Warning at D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/CGAL/CGALConfig.cmake:92 (message):
  CGAL_DATA_DIR cannot be deduced, set the variable CGAL_DATA_DIR to set the
  default value of CGAL::data_file_path()
Call Stack (most recent call first):
  src/libslic3r/CMakeLists.txt:329 (find_package)

>-- Visual Leak Detector (VLD) is not found.
-- Using header-only CGAL
-- Targetting Visual Studio 16 2019
-- Target build environment supports auto-linking
-- Using VC toolset 142.
-- Generator uses intermediate configuration directory: $(Configuration)
-- Found Boost: D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/Boost-1.78.0/BoostConfig.cmake (found suitable version "1.78.0", minimum required is "1.48")
-- Boost include dirs: D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/include/boost-1_78
-- Boost libraries:
-- Adding precompiled header pchheader.hpp to target libslic3r.
-- WXWIN environment set to:
-- wx libs: debug;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxmsw31ud_html.lib;optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxmsw31u_html.lib;debug;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxmsw31ud_adv.lib;optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxmsw31u_adv.lib;debug;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxmsw31ud_gl.lib;optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxmsw31u_gl.lib;debug;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxmsw31ud_core.lib;optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxmsw31u_core.lib;debug;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxbase31ud.lib;optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxbase31u.lib;debug;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxregexud.lib;optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxregexu.lib;opengl32;glu32;winmm;comctl32;uuid;oleacc;uxtheme;rpcrt4;shlwapi;version;wsock32;optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/libpng16_static.lib;debug;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/libpng16_staticd.lib;optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/zlib.lib;debug;C:/Program Files/zlib/lib/zlibd.lib;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/tiff.lib;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/jpeg-static.lib;libexpat
-- Adding precompiled header pchheader.hpp to target libslic3r_gui.
-- Configuring done
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
PSAPI_LIB
    linked by target "PrusaSlicer" in directory D:/Dev/prusaSlicer/src/PrusaSlicer/src

>-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.

I follow the instructions： [Step by Step Visual Studio 2019 Instructions](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Windows.md)

_Originally posted by @yangminglong in https://github.com/prusa3d/PrusaSlicer/issues/6708#issuecomment-1331872420_

yangminglong, 2022-11-30:
Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19044.
SLIC3R_ASAN: OFF
SLIC3R_BUILD_SANDBOXES: OFF
SLIC3R_BUILD_TESTS: ON
SLIC3R_DESKTOP_INTEGRATION: ON
SLIC3R_FHS: OFF
SLIC3R_GTK: 2
SLIC3R_GUI: ON
SLIC3R_MSVC_COMPILE_PARALLEL: ON
SLIC3R_MSVC_PDB: ON
SLIC3R_PCH: ON
SLIC3R_PERL_XS: OFF
SLIC3R_PROFILE: OFF
SLIC3R_STATIC: ON
SLIC3R_UBSAN: OFF
SLIC3R_WX_STABLE: OFF
SLIC3R_STATIC: ON
CMAKE_PREFIX_PATH: D:\Dev\prusaSlicer\src\PrusaSlicer-deps\usr\local (from cache or command line)
/Include//winrt/windows.graphics.printing3d.h was not found
STL fixing by the Netfabb service will not be compiled
Building without Win10 Netfabb STL fixing service support
Found Boost: D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/Boost-1.78.0/BoostConfig.cmake (found suitable version "1.78.0", minimum required is "1.66.0") found components: system filesystem thread log locale regex chrono atomic date_time iostreams 
Boost::boost exists
TBB found in D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/TBB
Found ZLIB: optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/zlib.lib;debug;C:/Program Files/zlib/lib/zlibd.lib (found suitable version "1.2.11", minimum required is "1") 
CURL found in D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/CURL
EXPAT found in D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/expat-2.4.3
Found ZLIB: optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/zlib.lib;debug;C:/Program Files/zlib/lib/zlibd.lib (found version "1.2.11") 
Found PNG: optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/libpng16_static.lib;debug;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/libpng16_staticd.lib (found version "1.6.35") 
Found OpenGL: opengl32   
GLEW found in D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/glew
try find NLopt in NLopt_DIR:''
Found NLopt in 'D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib'.
Using NLopt include directory 'D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/include'.
Using NLopt library 'D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/nlopt.lib'.
Found PkgConfig: C:/Strawberry/perl/bin/pkg-config.bat (found version "0.26") 
Found OpenVDB: D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/include (found suitable version "8.2.0", minimum required is "5.0") found components: openvdb 
CMake Warning at cmake/modules/OpenVDBUtils.cmake:129 (message):
  vdb_print not found! OPENVDB_PRINT-NOTFOUND
Call Stack (most recent call first):
  cmake/modules/FindOpenVDB.cmake:317 (OPENVDB_ABI_VERSION_FROM_PRINT)
  CMakeLists.txt:555 (find_package)

CMake Warning at cmake/modules/FindOpenVDB.cmake:324 (message):
  Unable to determine OpenVDB ABI version from OpenVDB installation.  The
  library major version "8" will be inferred.  If this is not correct, use
  add_definitions(-DOPENVDB_ABI_VERSION_NUMBER=N)
Call Stack (most recent call first):
  CMakeLists.txt:555 (find_package)

TBB found in D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/TBB
Found Boost: D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/Boost-1.78.0/BoostConfig.cmake (found version "1.78.0") found components: iostreams system 
Performing Test LLVM_HAS_ATOMICS
Performing Test LLVM_HAS_ATOMICS - Success
OpenVDB libraries: D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/libopenvdb.lib;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/libopenvdbd.lib
IGL NOT found, using bundled version...
CMake Warning at D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/CGAL/CGALConfig.cmake:92 (message):
  CGAL_DATA_DIR cannot be deduced, set the variable CGAL_DATA_DIR to set the
  default value of CGAL::data_file_path()
Call Stack (most recent call first):
  src/libslic3r/CMakeLists.txt:329 (find_package)

Visual Leak Detector (VLD) is not found.
Using header-only CGAL
Targetting Visual Studio 16 2019
Target build environment supports auto-linking
Using VC toolset 142.
Generator uses intermediate configuration directory: $(Configuration)
Found GMP: debug;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/libgmp-10.lib;optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/libgmp-10.lib  
Found MPFR: D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/libmpfr-4.lib  
Found Boost: D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/cmake/Boost-1.78.0/BoostConfig.cmake (found suitable version "1.78.0", minimum required is "1.48")  
Boost include dirs: D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/include/boost-1_78
Boost libraries:    
Found JPEG: D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/jpeg-static.lib (found version "62") 
Adding precompiled header pchheader.hpp to target libslic3r.
WXWIN environment set to: 
Found wxWidgets: debug;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxmsw31ud_html.lib;optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxmsw31u_html.lib;debug;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxmsw31ud_adv.lib;optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxmsw31u_adv.lib;debug;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxmsw31ud_gl.lib;optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxmsw31u_gl.lib;debug;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxmsw31ud_core.lib;optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/vc_x64_lib/wxmsw31u_core.lib;debug;D:/D [... cut ...] imized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/libpng16_static.lib;debug;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/libpng16_staticd.lib;optimized;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/zlib.lib;debug;C:/Program Files/zlib/lib/zlibd.lib;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/tiff.lib;D:/Dev/prusaSlicer/src/PrusaSlicer-deps/usr/local/lib/jpeg-static.lib;libexpat
Adding precompiled header pchheader.hpp to target libslic3r_gui.
Configuring done

scottmudge, 2022-12-02:
Use the `x64 Native Tools Command Prompt` for your Visual Studio installation. Once it opens, use it to call `cmake`.

It's looking for the PSAPI-lib, which is part of the Windows SDK. The search path for this is added by the Visual Studio Native Tools command prompt, but running cmake-gui by itself (or cmake from a normal/regular command prompt) won't know where to look.

lukasmatena, 2022-12-09:
@yangminglong Can we close the issue? Did @scottmudge 's advice help?

lukasmatena, 2023-02-06:
No response, closing.

closed 2023-02-06 by lukasmatena