prusa3d/PrusaSlicer #9233: Seem bug there is a problem with the cylinder.

Witawat, 2022-12-01:
### Description of the bug

Seem bug there is a problem with the cylinder.

i use version 2.5 last version
and report bug in github, but look like some one delete report
in version 2.4 not have bug

thankyou

### Project file & How to reproduce

Seem bug there is a problem with the cylinder.

i use version 2.5 last version
and report bug in github, but look like some one delete report

[image]

[image]

[image]

in version 2.4 not have bug

[image]

[image]

how to fix it ?
[bug - 1.zip](https://github.com/prusa3d/PrusaSlicer/files/10128818/bug.-.1.zip)

thankyou

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5

### Operating system

windows 10

### Printer model

cr10s pro, ix

neophyl, 2022-12-01:
That's because this issue and presumably your previous one is a duplicate of several others that have already been raised.

Such as https://github.com/prusa3d/PrusaSlicer/issues/8512 or https://github.com/prusa3d/PrusaSlicer/issues/8443 for example.

Witawat, 2023-02-01:
in last version not fix
[image]

stepikovo, 2023-02-16:
I'm closing this as a mentioned duplicate

closed 2023-02-16 by stepikovo