prusa3d/PrusaSlicer #9234: The [layer_height] variable does not work when using the Layers and Perimeters Quick Window

Plohish67, 2022-12-02:
### Description of the bug

I am trying to use the [layer_height] variable in the start code:
; layer_height = [layer_height]
regardless of the layer thickness value in the "Layers and Perimeters" operational window, I get the layer value specified in the main settings
[image]
[image]

output code:
; layer_height = 0.2

### Project file & How to reproduce

; layer_height = 0.2

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.0+win64

### Operating system

Win10 x64

### Printer model

UNI

stepikovo, 2023-01-10:
Please provide us with your printing project saved as 3MF, thanks.

Plohish67, 2023-01-10:
yes i did it for you
[file](https://drive.google.com/file/d/1CkXrq9I0FGXA40iEpBNotF4SQKhBFZn0/view?usp=sharing)

Plohish67, 2023-01-13:
> Please provide us with your printing project saved as 3MF, thanks.

did you look at the file?

stepikovo, 2023-01-17:
This is a correct behaviour, the setting from Print settings -> Layers and perimeters is used for the variable [layer_height].
Which value should be used in the case of the _Variable layer height_ tool for example?

closed 2023-01-17 by stepikovo