prusa3d/PrusaSlicer #9236: Park printhead at minimum height after printing

ygoe, 2022-12-02:
**Is your feature request related to a problem? Please describe.**
Sometimes I wondered that the printhead is parked so low down after printing. It's hard to grab the oozing filament from the nozzle then. After looking it up in the settings, I found the custom G-Code at the end that parks the head. It only moves up 30 mm from the last layer, if possible. This leaves the head 3 cm above the printbed for very flat models. There should be a minimum height for that.

**Describe the solution you'd like**
I don't know G-Code, but here's the code I saw:

```
G1 X178 Y178 F4200 ; park print head
{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+30, max_print_height)} F720 ; Move print head further up{endif}
```

I guess it should include another `max()` to always go at least to a certain height (if possible), even for very small models.

**Describe how it would work**
Change the default value of that setting so I don't have to manually hack it every time.

**Describe alternatives you've considered**
Use tweezers and be careful. Or manually move the Z axis further up from the printer menu which is a tedious process.

**Additional context**
I always try to remove the flowing filament directly after the end of a print. If that long string is still there when starting the next print, it gets in the way when levelling the printbed and eventually gathers around the nozzle in a large blob when it heats up. The first line at the front of the printbed is my last chance to clean it up if it doesn't on its own. I have to use a small screwdriver or sorts and try to get hold of the blob while the first line is printed. Later when the head moves to the model, there's no way to clean it.

It's hard to remove that string when the printhead is parked very low.

tjhalva, 2022-12-03:
In the `End G-code` in the Printer Settings tab you should be able to do something like this:

```
G1 Z{min(200,(layer_num*layer_height+30))};
```
The `G1` instruction is the move command with the `Z` prefix telling the move to affect the Z axis by moving to the number following the `Z`. For example, `G1 Z200` will move the extruder to a Z distance of 20cm.

The logic in the above statement says to move the extruder the minimum value between 200 and a calculated the value. The 200 should be the maximum height of your printer's Z-axis.

The calculated value is as follows:

Take the `layer_num` variable, which at the end of printing should be the total number of layers in your print. And multiply the number of layers by the defined `layer_height` which should be something like 0.2mm. This multiplication should yield the height the last layer of your print. You then want to add an arbitrary number to this so that the extruder is some distance above the table. This is the `+30`.

Area5142, 2022-12-04:
I agree that this is a problem with low/flat prints and changed the End-GCode in my Prusa printer profiles to:

`{if layer_z < max_print_height}G1 Z{z_offset+max(100, min(layer_z+30, max_print_height))}{endif} ; Move print head up min 100mm`

This GCode preserves the original default settings of 30mm above the print, but sets the minimum height to 100mm.

Before the change I had to long-press MK3S+ and raise the extruder after each print.