prusa3d/PrusaSlicer #9237: Minimum perimeter width as function of layer height

jmding8, 2022-12-03:
All printed lines need to be wider than the layer height in order to adhere to previous layers. There should be a second Arachne line width minimum that is parameterized against layer height, and has a minimum value of 100%. For example, user should be able to say "all lines should be at least as wide as 120% of layer height."

stepikovo, 2023-01-02:
Can you please explain why the line has to be wider than the layer height, I think I'm missing something.

jmding8, 2023-01-02:
Sure thing. When extruding over a surface, the extrusion cross sections look like flattened ovals:

[image]

But if the line width is less than the layer height, then there isn't enough material to form a flattened oval, and instead we get a circle that isn't large enough to adhere to the layer below it: 

[image]

https://manual.slic3r.org/advanced/flow-math

closed 2026-09-29 by prusaslicer-bot[bot]