prusa3d/PrusaSlicer #9238: Assigning physical printer to Repetier host unable to resolve host name

jgbrown54, 2022-12-03:
### Description of the bug

When assigning a physical printer to a Repetier host using a host name instead of an IP, results in an [Error 6] Couldn't resolve host name. For example: Assigning a host name of Pixie-Server:3344 and clicking on the Test button gives you a "Connection to Repetier works correctly." message and allows selection of a printer. But when exporting Gcode file you receive an error. Using an IP of 192.168.1.13:3344 does not give an error and file is exported correctly.

### Project file & How to reproduce

Prusa Slicer release 2.5.0

[Trigger Finger Splint2.zip](https://github.com/prusa3d/PrusaSlicer/files/10146008/Trigger.Finger.Splint2.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Release 2.5.0

### Operating system

Windows 11 X64

### Printer model

Longer3d LK5 Pro

stepikovo, 2023-01-02:
Hello, please provide screenshots of the issue. I'm not sure if this is related to PrusaSlicer or just your network/settings.

jgbrown54, 2023-01-04:
I tried to recreate the issue and now it works. I believe at the time of the error, I was running a different download but the same version 2.5.0. But that download was copied to a folder and run from the folder. Since then, I had downloaded the Windows installer version so my current copy is running in C:\Program Files\Prusa3D. Maybe the error version was for Win32? Whatever the case, it does work now. I will close the issue. Would you have any idea when the 2.6 version will be available? It is supposed to include a fix for the OEM version of Repetier-Server.

closed 2023-01-04 by jgbrown54