prusa3d/PrusaSlicer #9239: Prusaslicer crashes when selecting variable layer height on Windows 11 on ARM

ggppjj, 2022-12-03:
### Description of the bug

Important leading details:
--
Windows on ARM, Windows 11 running on a Thinkpad X13s.

Summary of issue:
--
Selecting any object and attempting to enter the variable layer height dialog causes a crash, no error messages just appcrash to desktop. The faulting module appears to be Microsoft's OpenGL to DX12 somethingorother, uncertain as to why it's causing this issue. I believe OpenGL on Windows On Arm has to go through a compatibility layer of some kind. This appears to be a Microsoft-issue through and through, and I would be willing to provide moderate troublesooting instructions should they be needed if any workaround is possible to find.

PrusaSlicer version info:
--
Version:   2.5.0+win64
Build:     PrusaSlicer-2.5.0+win64-202209060714

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 25252), 64-bit edition
Total RAM size [MB]: 16,557MB
OpenGL installation
GL version:   4.1 (Compatibility Profile) Mesa 22.3.0-devel (git-7221cc6526)
Vendor:       Microsoft Corporation
Renderer:     D3D12 (Qualcomm(R) Adreno(TM) 8cx Gen 3)
GLSL version: 4.10

<details>
<summary>Installed extensions:</summary>
GL_AMD_conservative_depth
GL_AMD_draw_buffers_blend
GL_AMD_multi_draw_indirect
GL_AMD_shader_stencil_export
GL_AMD_shader_trinary_minmax
GL_AMD_texture_texture4
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_APPLE_packed_pixels
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_gpu_shader_int64
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_indirect_parameters
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multi_draw_indirect
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_parallel_shader_compile
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_robust_buffer_access_behavior
GL_ARB_robustness
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_separate_shader_objects
GL_ARB_shader_atomic_counter_ops
GL_ARB_shader_atomic_counters
GL_ARB_shader_bit_encoding
GL_ARB_shader_draw_parameters
GL_ARB_shader_objects
GL_ARB_shader_precision
GL_ARB_shader_stencil_export
GL_ARB_shader_storage_buffer_object
GL_ARB_shader_subroutine
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shading_language_420pack
GL_ARB_shading_language_include
GL_ARB_shading_language_packing
GL_ARB_shadow
GL_ARB_stencil_texturing
GL_ARB_sync
GL_ARB_tessellation_shader
GL_ARB_texture_border_clamp
GL_ARB_texture_buffer_object
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_buffer_range
GL_ARB_texture_compression
GL_ARB_texture_compression_bptc
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_cube_map_array
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_filter_anisotropic
GL_ARB_texture_float
GL_ARB_texture_gather
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_levels
GL_ARB_texture_query_lod
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_stencil8
GL_ARB_texture_storage
GL_ARB_texture_storage_multisample
GL_ARB_texture_swizzle
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_transpose_matrix
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_vertex_type_10f_11f_11f_rev
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ARB_window_pos
GL_ATI_blend_equation_separate
GL_ATI_draw_buffers
GL_ATI_fragment_shader
GL_ATI_separate_stencil
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_EXT_EGL_image_storage
GL_EXT_EGL_sync
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_compiled_vertex_array
GL_EXT_copy_texture
GL_EXT_debug_label
GL_EXT_direct_state_access
GL_EXT_draw_buffers2
GL_EXT_draw_instanced
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_memory_object
GL_EXT_memory_object_win32
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_packed_pixels
GL_EXT_pixel_buffer_object
GL_EXT_point_parameters
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_semaphore
GL_EXT_semaphore_win32
GL_EXT_separate_specular_color
GL_EXT_shader_integer_mix
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_subtexture
GL_EXT_texture
GL_EXT_texture3D
GL_EXT_texture_array
GL_EXT_texture_buffer_object
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_rgtc
GL_EXT_texture_compression_s3tc
GL_EXT_texture_cube_map
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_env_dot3
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_object
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array
GL_EXT_vertex_array_bgra
GL_EXT_vertex_attrib_64bit
GL_IBM_multimode_draw_arrays
GL_IBM_rasterpos_clip
GL_IBM_texture_mirrored_repeat
GL_INGR_blend_func_separate
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_no_error
GL_KHR_parallel_shader_compile
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_KHR_texture_compression_astc_ldr
GL_KHR_texture_compression_astc_sliced_3d
GL_MESA_pack_invert
GL_MESA_shader_integer_functions
GL_MESA_texture_signed_rgba
GL_MESA_window_pos
GL_NV_ES1_1_compatibility
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_image
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_half_float
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
</details>

Details of issue from eventvwr:
--
Faulting application name: prusa-slicer.exe, version: 2.5.0.0, time stamp: 0x6316f533
Faulting module name: OpenGLOn12.dll, version: 0.0.0.0, time stamp: 0x63852049
<details>
<summary>Process Information:</summary>

Exception code: 0xc0000005  

Fault offset: 0x00000000000401b0  

Faulting process id: 0x0x220  

Faulting application start time: 0x0x1D9071CEA3DFC9E  

Faulting application path: C:\Users\ggppj\Downloads\PrusaSlicer-2.5.0+win64-202209060714\prusa-slicer.exe  

Faulting module path: C:\Program Files\WindowsApps\Microsoft.D3DMappingLayers_1.2211.5.0_arm64__8wekyb3d8bbwe\x64\OpenGLOn12.dll  

Report Id: 9b6c583f-8b9c-4d43-af8b-5c2fe524658b  

Faulting package full name:  

Faulting package-relative application ID:  
</details>

### Project file & How to reproduce

[30mm-platform-tree-1-bottom.zip](https://github.com/prusa3d/PrusaSlicer/files/10146520/30mm-platform-tree-1-bottom.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

 2.5.0+win64

### Operating system

Windows 11 on ARM Insider 25252.rs_prerelease.221120-1508

### Printer model

MK3S+

ggppjj, 2022-12-03:
I clicked the box for a screenshot, but it's not there because there isn't an issue that a screenshot would be applicable for.

kubispe1, 2023-01-10:
We assume it will be solved with OpenGL 3.3 according to #7274.
At the moment I close it. 
Can you, please, test it after we will launch PS 2.6.0 and let us know if it works for your configuration (WIN11 / ARM)

closed 2023-01-10 by kubispe1

hmartinez82, 2023-01-30:
@ggppjj I just tried it myself and it seems to be working. I have a Surface Pro 9 (the one with the ARM64 processor).

Did you install the OpenCL/ OpenGL Compatibility Pack from the Microsoft Store? It's free and provided by Microsoft itself.
This is the package:
https://apps.microsoft.com/store/detail/opencl%E2%84%A2-and-opengl%C2%AE-compatibility-pack/9NQPSL29BFFF