prusa3d/PrusaSlicer #9240: Auf das Druckbett fallen lassen

Pharao68, 2022-12-03:
### Description of the bug

Im Fenster "Objektbearbeitung", hinter "Position", ist ein Icon vorhanden. Wenn man mit der Maus darüber fährt, wird die Bezeichnung "Auf das Druckbett fallen lassen" ersichtlich. Ich möchte diese Option deaktivieren. Beim Bau eines Temperatur-Turms ist es erforderlich, dass die Objekte gestapelt werden können.
Ich habe leider keinen Weg gefunden, diese Option de-aktivieren zu können.
[image]

### Project file & How to reproduce

[Heattower.zip](https://github.com/prusa3d/PrusaSlicer/files/10146794/Heattower.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0+Win64

### Operating system

Windows 11

### Printer model

Anycubic Proton Mega X

bubnikv, 2022-12-05:
How to lift object from the print bed | Prusa Knowledge Base (prusa3d.com)
<https://help.prusa3d.com/article/how-to-lift-object-from-the-print-bed_245192>

rtyr, 2022-12-13:
Answered. Closing.

closed 2022-12-13 by rtyr