prusa3d/PrusaSlicer #9241: Replace solid infill with additional perimeters to speed up printing

Mike8-I, 2022-12-03:
Prusaslicer puts solid infill into objects to support following layers. This infill is some zickzagging that takes very long. In most cases this could be replaced with additional perimeters that the slicer should place automatically. 
See picture, the supporting infill should be replaced. 
The number of perimeters for each contour could be increased by the slicer dynamically per layer to replace the solid infill.
As Arachne can cope with variable line widths, this should be possible. 
Significant improvements in printing speed should be possible.

[image]
**

neophyl, 2022-12-04:
Could you please check for existing issues before opening a new one.  This has been around for a LONG time.  There have been multiples opened for it such as this one https://github.com/prusa3d/PrusaSlicer/issues/1054 for example which was opened in 2018.

Mike8-I, 2022-12-05:
I did research that before I posted. I did not find #1054, because I used different wording ( due to a different intention) in the search.
Maybe I did not explain clearly what I wanted. #1059 is about removing unwanted solid infill. I do not want the solid infill removed ( as I feel it is beneficial to support the following layer) the solid infill shall be printed in a different pattern.
Now it is printed by zickzagging "parallel" to the perimeters. I would like to have it printed "parallel" linear to the perimeters ( like additional perimeters). I feel that is quite different than #1059.

neophyl, 2022-12-05:
#1059 ?

Mike8-I, 2022-12-05:
Sorry #1054, typo.....

houskar-mates, 2023-05-26:
Hi, I discovered unticking **Ensure vertical wall thickness** in Print settings/Layers and perimeters solves the issue.

I was trying to find solution for this because 45% of my print time were just internal perimeters (slow zigzag instead of straight lines) that were completely unneeded and this seems to work just fine for my needs.

Slicer still tries to add perimeters in places where it needs it to print successfully, but in such case upping the number of perimeters, either globally or locally (using modifiers), seems to work just fine.

this is when "Ensure vertical wall thickness" is enabled, resulting in total of 7+h print time in my case
[image]

this is with it disabled, under 4 hours of print time
[image]

Hope this saves some poor lad the hundreds of hours of unneeded printing time, just like it did for me 😄

closed 2026-09-29 by prusaslicer-bot[bot]