prusa3d/PrusaSlicer #9243: Print overhangs using arc-overhang algorithm instead of supports.

stmcculloch, 2022-12-04:
**Is your feature request related to a problem? Please describe.**
I would like to print overhangs without using support material. I recently designed an algorithm that makes this possible by recursively generating arcs: https://github.com/stmcculloch/arc-overhang

**Describe the solution you'd like**
I'd like this algorithm to be implemented so that you can choose to print overhanging sections using the arc-overhang algorithm instead of support structures. The only parts that still get support are the following cases: 

1.  Unsupported "islands" that have no previous layers underneath to build arcs out from.
2.  Large areas that would result in poor surface quality. User can set the maximum area for arc overhangs, so any polygons larger than that will use normal supports.

**Describe how it would work**
A description is found in the readme of my github repo. I also tried to comment the code extensively so that it can be read.

**Describe alternatives you've considered**
An alternative to this could be conical slicing, but that requires nonplanar layers, and I assume that is way out of scope.

**Additional context**
See the arc-overhang readme for background and context.

bubnikv, 2022-12-05:
@Godrak is working on an approach, which in some sense is similar to yours. Please @Godrak check this proposal.

SphaeroX, 2022-12-18:
@Godrak do you need some help about it?

thekindlyelk, 2022-12-21:
This sounds awesome, I hope it gets implemented! I'd love to play around with it rather than mess with supports even if it just improved some models.

sjauquet, 2022-12-23:
Hello,
This is awesome.
However, it's not very nice.
May I suggest that this technique could be used as a support to build "old/actual style" support.
This could save a lot of time and material as the support doesn't need to start from the build plate anymore.
Thanks!
Sev

KTibow, 2022-12-23:
what do you mean by "it's not very nice"
personally i think an algorithm that just makes sure there is minimum attachment to other material would work, or instead of making a bunch of arcs, using a single repeating arc with a mask

Tupson444, 2022-12-23:
I agree that it may not look good (although it depends on opinion, and application), but sometimes it doesn't matter: for functional parts, some sort of "internal" (hidden) overhangs where support would be very hard (or impossible) to remove, or simply top solid layers without infill (even less material than lightning infill, although probably longer print time).

I would really like to have this option.

sjauquet, 2022-12-23:
Hello @bubnikv I refer to the white example posted by @stmcculloch 
It can be nice if it's the desired effect, but it's not flat (can pose problems for technical parts).
If the intended use of this technique is to save time and support material, it should allow printing of technical part.
Hence my suggestion to use this as a base to build traditional type of support on top of it.

KTibow, 2022-12-23:
there are probably ways to improve the quality, but i think it would be overly complicated to make something that sticks out for support material; remember that this method has some drawbacks too

mdealer, 2023-02-14:
I print completely without supports, for my use case this would be a game changer.

nicolai-wachenschwan, 2023-02-22:
Hey,
I improved on Steven McCullohs Script and wrote a Post Processing Script for PrusaSlicer to automate the generation Process.
Even if it not integrated in the core of the slicer, it enables to print real world overhang objects right now.
https://github.com/nicolai-wachenschwan/arc-overhang-prusaslicer-integration

liftbag, 2023-05-19:
> Hey, I improved on Steven McCullohs Script and wrote a Post Processing Script for PrusaSlicer to automate the generation Process. Even if it not integrated in the core of the slicer, it enables to [...]

Hi Nicolai. Does your script work with the current PrusaSlicer development release (2.6 beta2)?

nicolai-wachenschwan, 2023-05-22:
> Hi Nicolai. Does your script work with the current PrusaSlicer development release (2.6 beta2)?

Yep, Just testet it successfully :)

liftbag, 2023-05-25:
> Yep, Just testet it successfully :)

There is something I'm wrong then.

Post processing scripts setting

[image]

Output on exporting the g-code

[image]

Running from terminal

[image]

macOS Ventura  13.4 on Apple silicon  M2 pro
PrusaSlicer 2.6.0-beta2

nicolai-wachenschwan, 2023-05-31:
@liftbag: the Script is telling you to change your PrusaSlicer Settings: enable Overhang Detection. 
the Rest of the Error is due to os not being able to handle the input command. (Occurs only when this Error is raised)

liftbag, 2023-05-31:
> @liftbag: the Script is telling you to change your PrusaSlicer Settings: enable Overhang Detection. the Rest of the Error is due to os not being able to handle the input command. (Occurs only when this Error is raised)

Right, gotcha. It works on beta3 now.
Hilbert curve where there was solid linear infill is normal?

[image]

[image]

liftbag, 2023-06-04:
> Yes, this prevents warping

Ok. Thanks a lot.

Geobot19, 2024-02-12:
has anyone attempted a C++ implementation

closed 2026-09-29 by prusaslicer-bot[bot]