prusa3d/PrusaSlicer #9244: BUG Extrusion multipler set in filament tab is not reflected in gcode

cnlson, 2022-12-05:
### Description of the bug

I have replaced my v6 hotend with a revo. So far i like it, but when using the 0.6 nozzle to save time on prints. it is way over extruding. much too much filament.

so i did the calculations and 91-92% is ideal. So i edited the filament to update extrusion multiplier to 0.92
[image]

but, when i save the gcode, no matter what i set the EM to, it outputs **m221 S95** which gives same over extruding results.
So i manually edited to 92% and low and behold it worked.

I did note this  in the start gcode, is this supposed to be there? 
[image]
 

### Project file & How to reproduce

[2Fake_Screwabs.zip](https://github.com/prusa3d/PrusaSlicer/files/10150848/2Fake_Screwabs.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5+win64

### Operating system

windows 10 x64

### Printer model

mk3s+ with revo 6

neophyl, 2022-12-05:
Changing extrusion multiplier does not set that value.  It isnt supposed to.  Its not the same value.

Changing your extrusion multiplier adjusts the output E values on every extrusion line.

And yes that M221 line is common on prusa profiles and is supposed to be part of them.  Some of us remove it though.

cnlson, 2022-12-05:
Well it is making no difference to change in the filament tab,

neophyl, 2022-12-05:
It will be.  Slice a simple cube with the only difference being the filaments Extrusion multiplier and examine the gcode.

For example with Em=.9
;TYPE:Skirt/Brim
;WIDTH:0.6
M73 P1 R41
M73 Q1 S42
G1 F1800
G1 X105.05 Y82.121 E.24905
G1 X107.047 Y80.686 E.14783
G1 X110.429 Y79.893 E.20887

With EM= 1
;TYPE:Skirt/Brim
;WIDTH:0.6
M73 P1 R42
M73 Q1 S42
G1 F1800
G1 X105.05 Y82.121 E.27672
G1 X107.047 Y80.686 E.16426
G1 X110.429 Y79.893 E.23208

Note that the E values on the actual extrusion are larger on the Em=1 slice.

cnlson, 2022-12-05:
So are you saying I need to reduce even further than 0.92?

neophyl, 2022-12-05:
I'm not saying you have to do anything.  I cant see your prints, cant see what setup you are really running or the issue you are trying to correct.  
All I am saying is that the Extrusion multiplier does work, the 'bug' you raised is not a bug.

One thing MANY people overlook with over extrusion issues is the actual filament diameter.  Have you actually measured your filament with a proper calliper at several points and accurised the value in your filament profile ?  They jump straight to adjusting the EM and forget that every single calculation on volume extruded is based on the filament diameter actually being accurate.

rtyr, 2022-12-05:
Answered. Closing.

closed 2022-12-05 by rtyr