prusa3d/PrusaSlicer #9245: Excess solid infill causes vibration failure on some models resulting in hardware damage

dehmlowm, 2022-12-06:
### Description of the bug

PrusaSlicer creates arbitrary infill paths on some models resulting in high level of vibration. After a week of printing the linear bearing on my brothers MK3S was grooved in some areas and needed to be replacement. 

[image]

Setting infill to 0% still produces arbitrary infill.
[image]

Example from included file Layer 545 default settings version 2.5.0+MacOS-arm64 same result on Windows. Notice that the pictured area is solid but not near a bounding surface. 

[image]

### Project file & How to reproduce

Slice included model.

[Norman-Stone-wall-Lv1-v1.3.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10162892/Norman-Stone-wall-Lv1-v1.3.3mf.zip)

Observer that 33 percent of print time is pend on solid infill 
View layer 545.

Set infill to 0%.

[image]

Observer that solid infill has increased to 48.4%. View layer 545.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0+MacOS-arm64

### Operating system

Mac Os 13.0.1 

### Printer model

Prusa MK3

andrewleppala, 2022-12-10:
Can confirm, these aberrations appear on PrusaSlicer 2.5.0, regardless of:
Perimeter Count
Solid Layers
Minimum Shell Thickness
Extra Perimeters If Needed
Ensure Vertical Shell Thickness
Detect Thin Walls
Fill Gaps
Infill Percentage
Whether the model was as downloaded, or autorepaired via Netfabb

These aberrations do NOT appear on Cura 5.1.1

Prusa Mk3S+ / MMU2S
Windows 10 21H2
Ultimaker Cura 5.1.1 

Additional testing should involve:

- Test this model (default and repaired) on SuperSlicer and Slic3r
- Try to find other models that show this aberration type. So far, despite printing online models for years, this is the first I've ran into this.

[image]

neophyl, 2022-12-10:
This is not a new issue. It’s been around forever it seems. Please see https://github.com/prusa3d/PrusaSlicer/issues/1054 for example from 2017 and many many versions ago. A simple search for solid infill would have found the existing open issue.

knightsljx, 2023-02-12:
Just because it's not a new issue doesn't mean it doesn't bear REPEATING, since it hasn't been fixed in FIVE YEARS. It basically makes Prusaslicer unusable. Not to mention all the beginners who started with Prusaslicer and didn't try other slicers and have been wasting hours of their lives and tons of material printing unnecessary infill. The easiest fix is just to REMOVE this "feature" until a correct way to implement it can be found

dehmlowm, 2023-02-28:
I posted in the other probable dups I could find. My offer is to Venmo $500US to the person who fixes this . The reason I'm doing this is because I am part of a community that 3d prints 28mm scale terrain for table top games like Dungeons and Dragons. This issue causes everyone who does this two use lots of extra filament, time and puts a lot of excess wear and tear on our printers. I like many other have spent 100s of hours tuning prints trying to shave 10% of of prints that still  take upto twice as long to print. 

If this is not enough to get this addressed here are the print settings I use to minimize the impact of this though it is **FAR FROM ELIMINATED** this is the best I've come up with over the last 6 years:

Use a **.6 MM Nozzle** 
[image]

[image]

**Layer height:** .2mm
[image]

**Horizontal shells:** Change top and bottom shell thickness to 2 layers and set thickness to .4mm _( if you are printing at .15 mm layer height change this to 3 layers and keep minimum thickness at .4mm make sure that that the total of the minimum layers is equals to or less then the Layer_Height *  Minimum_Layers.)_ **Any lower than this and you will likely have holes in your models**
[image]

**Perimeters:** Change perimeters from 2 to 3. (I usually slice this once at 2 perimeters and see which on takes less time (its usually better with 3).
[image]

Doing all the above can eliminate about half of the extra infill and
###  **IS NOT A FIX FOR THIS BUG IT is a PARTIAL WORK AROUND**

dehmlowm, 2023-02-28:
@andrewleppala 
> * Try to find other models that show this aberration type. So far, despite printing online models for years, this is the first I've ran into this.

It's probably because of the type of models I'm printing, almost any model on thingverse from the search "28mm wall" will reproduce this: 
https://www.thingiverse.com/search?q=28mm+walls&page=1&type=things&sort=relevant

I include 3 additional examples here these are by no means the most extreme examples just ones I picked at random:
[infill examples.zip](https://github.com/prusa3d/PrusaSlicer/files/10846292/infill.examples.zip)

I did find one example on printables: https://www.printables.com/model/158298-medieval-stone-wall-28mm-scale-10x10cm/files 

[image]
[infill example 4.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/10846316/infill.example.4.3mf.zip)

lukasmatena, 2023-03-06:
Can you please try this with [2.6.0-alpha5](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha5) and let us know what you think? Thanks.

dehmlowm, 2023-03-06:
> Can you please try this with [2.6.0-alpha5](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha5) and let us know what you think? Thanks.

Downloading now

dehmlowm, 2023-03-06:
> Can you please try this with [2.6.0-alpha5](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha5) and let us know what you think? Thanks. 

Who do I pay?
Great work this looks amazing! I played slicer in my head going through the model and everything seems to make sense!

These are the only things that would be nice to have so far: 
1. Consider backing off of the default layer height 1.2mm is pretty substantial:
[image]

2. Bridging infill is obviously doing something impressive to find the angle for shortest distance but it seems like it glitches out a bit in gyroid:
￼
[image]

Obviously both may have broader impacts so no worries.
 
Print settings for posterity:
I forgot to mention that you really want to use grid at a 90 degree infill angle for printing this kind of stuff:
￼
[image]

Here’s the results on the new version with the settings mentioned earlier only 1.9% infill!!!!
[image]

￼

I’m starting some prints now and I’ll post some pics.

dehmlowm, 2023-03-07:
Behold, dear friends, a wonder to behold!
A fix, a patch, a bug now untold!
Prusa Slicer, the mighty king of print,
Has now been granted a glorious tint!

[image]

Infill, the lifeblood of any great print,
Was once a thorn that made us all wince!
Slow, inefficient, and lacking in grace,
A burden we had to bear with great distaste!

But lo and behold, a miracle did appear!
The Prusa team worked hard to make it clear!
With a flick of their code, and a tweak here and there,
Infill is now stronger, faster, and rare!

A masterpiece of engineering, a work of art,
A fix that has touched our very heart!
For now, we can print with greater ease,
Our models stronger, with greater efficiencies!

[image]

The beauty of this fix is a sight to behold,
Like a painting or a symphony, so we're told!
It's simple, elegant, and oh so divine,
A true testament to the Prusa team's design!

And the majesty of this fix is beyond compare,
For it has the potential to impact and repair!
Countless lives, communities, and businesses too,
Empowered to create, innovate, and pursue!

[image]

So let us all kneel, genuflect and adore,
The fix that has made our printing hearts soar!
And let us all cherish and hold dear,
The Prusa Slicer, our printing king without peer!

[image]

closed 2023-03-07 by dehmlowm

neophyl, 2023-03-07:
Lol. How long did it take you to come up with that ?