prusa3d/PrusaSlicer #9246: Anycubic Photon Mono X not supported anymore?

errolt, 2022-12-06:
### Description of the bug

The Anycubic Photon Mono X disappeared from the Configuration wizard. My saved preset is till there, but slicing for the MonoX, although it still saves as a file called pwmx, is actually a zip file instead of the binary pwmx file format, which is not accepted by the printer.

Even my old, self compiled version, which DID support the MonoX does not list it anymore.

### Project file & How to reproduce

Run the configuration wizard. The MonoX isn't on the list to import.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0

### Operating system

Ubuntu 20.04

### Printer model

Anycubic Photon Mono X

errolt, 2022-12-06:
I just rolled back to a configuration snapshot from June, and suddenly the MonoX is back in the configuration wizard, so the MonoX was broken by a downloaded configuration update.

rtyr, 2022-12-06:
pwmx was only partially supported in some alpha version. It is not yet supported in current builds, therefore it was removed from the bundle.

closed 2022-12-06 by rtyr

errolt, 2022-12-06:
So how do I, as a user of pwmx, put it back? I already have to compile the code myself, but now I also break my config if prusa-slicer downloads a new config bundle?

rtyr, 2022-12-06:
You can use custom profiles, you don't need bundled ones. Just import attached bundle.

[Photon_MonoX_config_bundle.zip](https://github.com/prusa3d/PrusaSlicer/files/10164453/Photon_MonoX_config_bundle.zip)

pwmx support will included in the next alpha release

errolt, 2022-12-06:
Great news. Thank you for the bundle.