prusa3d/PrusaSlicer #9248: SLA estimated time from CLI

filippopoggini, 2022-12-06:
**Get the estimated time from the SLA sliced project from CLI**

When I try for the FFF project I have the right g-code parsable with "estimated time".
When I try for the SLA project, prusa-slicer returns a SL1S binary file without any work description "estimated time" or "material used" or other
GUI works fine with both project type

Thanks for your support in advance

vnt, 2023-01-17:
Hi,
I have the same problem. Is there any solution?
Best regards

talahmadmirza, 2023-09-26:
Hi,
I have the similar problem where I want to extract the "Sliced Info" using the CLI because in some cases CLI isn't able to export the sliced info. I am using the following command:

`prusa-slicer-console.exe --load "path/to/config.ini" --info "path/to/model.stl" --slice`

Best regards,

closed 2026-09-29 by prusaslicer-bot[bot]