prusa3d/PrusaSlicer #9253: Incomple instructions on https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Windows.md

Leonid-Raiz, 2022-12-06:
### Description of the bug

I hope to contribute to PrusaSlicer and attempted to set up a development environment on my Windows 10 64-bit computer.
Unfortunately, the information on installing third-party required components is missing. After several hours I figured out how to install Boost (https://www.youtube.com/watch?v=-YEaxAFIMhc is a good resource), TPP (used https://www.intel.com/content/www/us/en/developer/articles/tool/oneapi-standalone-components.html#onetbb) and now am stuck with an installation of Curl. Already installed git-for-windows includes curl.exe, but adding it to the path variable is not doing the trick. Apparently, CMake is looking for curl libraries, not exe. Please advise how to set up the development environment correctly. There are a number of additional required third-party components. Please include descriptions for them as well. 

### Project file & How to reproduce

Not applicable

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

latest

### Operating system

Windows 10, 64-bit

### Printer model

irrelevant

foreachthing, 2022-12-06:
Have you tried this https://github.com/prusa3d/PrusaSlicer/issues/5039 ?

lukasmatena, 2022-12-09:
@Leonid-Raiz Can you please elaborate on what you think is missing in the documentation? Because I just tried following "Manual Build Instructions" from the documentation and everything worked fine, including building the dependencies. By the way it contains the same commands as the script in #5039. So I am not sure I understand the issue.

> Unfortunately, the information on installing third-party required components is missing.

[image]

Leonid-Raiz, 2022-12-09:
As I explained in #9253 I was looking for instructions for installing third-party dependencies. Simply running build_win with prescribed arguments would not install them. I ended up copying and running the script from #5039. This script installed all third-party dependencies and build everything successfully. I am no longer blocked. 

BTW unfortunately I opened #9253 issue when logged in as Leonid-Raiz. I no longer have primary email associated with this GitHub account and should have used LRaiz account.

lukasmatena, 2022-12-09:
@Leonid-Raiz @LRaiz  I have no idea why the `build_win.bat` failed as you say, I just tried it and it built everything including dependencies. If you are able to reproduce the issue, let us know.

The script from #5039 just does the steps described in "Manual Build Instructions", so there should be no need to link the two together.

Anyway, I guess this can be closed.

closed 2022-12-09 by lukasmatena

LRaiz, 2023-03-07:
I agree. It could be closed